libretro/Lakka-LibreELEC #360: Cubietruck booting process stucked

prokhyon, 2018-01-31:
What I've experienced:

Lakka-Allwinner.Cubietruck.arm-2.1.img:
- The image simply does not boot. No error message on screen, it's just not starting. I've tried to write the image multiple times to SD card and multiple displays with HDMI, d-sub etc., the result is the same.
- When writing the image to SD card, the Etcher (and Win32DiskImager) can't unmount the drive after finishing the process, shows an error, which does not appear when writing any other non-Lakka image to the same SD card. I've downloaded this image multiple times, the image is not corrupted.

Lakka-Allwinner.Cubietruck.arm-2.1-rc6.img:
- Same unmounting issue here as described above after writing the image to SD card.
- This image boots, but it can't finish the boot process. The first two lines of dmesg are:
```
[FAILED] Failed to start Load Kernel Modules.
See 'systemtl status sstemd-modules-load.service' for details.
```
But the shell is not appearing, the boot time dmesg is stucked, I can't type any command. According to dmesg the booting process ends at different points of the booting sequence.

My Hardware:
- Cubietruck (first edition) with Allwinner A20 cpu
- display: LG 49LJ59 TV, HDMI input
- sd card: Kingston SDC10G2/16GBSP

kivutar, 2018-01-31:
Related to https://github.com/libretro/Lakka-LibreELEC/issues/307

ToKe79, 2018-01-31:
Closing this issue, as it is duplicate, please follow #307 for further development. Thank you!

closed 2018-01-31 by ToKe79