libretro/Lakka-LibreELEC #363: Odroid C0 network issues

jordanworner, 2018-02-01:
- Which version of Lakka are you using?
2.1 Odroid C1 image
- What system hardware are you using?
Odroid C0 with RTL8188CUS usb wifi dongle
- What did you do?
Try to scan for wifi in Lakka
- What did you expect to happen?
Display available networks
- What happened instead?
Nothing

On boot and shutdown the Connection service takes a long time to start and stop, 5+ mins. I'm assuming this is probably due to the Odroid C0 not actually having an ethernet port. If I access the command line directly and run anything network related (ifconfig etc) it just hangs. lsusb shows the wifi dongle.

Is there anything I can try disabling/changing so I can use the wifi and remove the long boot and shutdown?

Thank you in advance.

Jerome-PS, 2019-03-23:
Hi,
I have a similar issue on PC (x86_64).
I plug in a RTL8188CUS dongle and it is not recognized.
lsusb gives:
Bus 001 Device 004: ID 2001:3308 D-Link Corp. DWA-121 802.11n Wireless N 150 Pico Adapter [Realtek RTL8188CUS]
dmesg gives:
[  230.524673] usb 1-5: new high-speed USB device number 4 using ehci-pci

But if I plug the same USB dongle in a Ubuntu laptop, I get a more verbose dmesg:
[  262.760043] usb 2-1: new high-speed USB device number 2 using ehci-pci
[  262.917772] usb 2-1: New USB device found, idVendor=2001, idProduct=3308
[  262.917777] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  262.917781] usb 2-1: Product: 802.11n WLAN Adapter
[  262.917785] usb 2-1: Manufacturer: Realtek
[  262.917788] usb 2-1: SerialNumber: 00e04c000001
[  263.303183] rtl8192cu: Chip version 0x10
[  263.388931] rtl8192cu: Board Type 0
[  263.389171] rtl_usb: rx_max_size 15360, rx_urb_num 8, in_ep 1
[  263.389286] rtl8192cu: Loading firmware rtlwifi/rtl8192cufw_TMSC.bin
[  263.396877] ieee80211 phy1: Selected rate control algorithm 'rtl_rc'
[  263.411098] usbcore: registered new interface driver rtl8192cu
[  263.462249] usbcore: registered new interface driver rtl8xxxu
[  263.532348] rtl8192cu 2-1:1.0 wlxc8be19040034: renamed from wlan1
[  263.590271] IPv6: ADDRCONF(NETDEV_UP): wlxc8be19040034: link is not ready
[  263.594319] rtl8192cu: MAC auto ON okay!
[  263.629568] rtl8192cu: Tx queue select: 0x05
[  264.178832] IPv6: ADDRCONF(NETDEV_UP): wlxc8be19040034: link is not ready
[  264.241381] IPv6: ADDRCONF(NETDEV_UP): wlxc8be19040034: link is not ready

Also the dongle's blue LED lights up on Ubuntu, but not on lakka.

So I think the driver for this USB dongle is not compiled, either in the kernel or as a module (I don't know much about the lakka distro).

jordanworner, 2019-07-28:
I was recently playing around with the Odroid C0 again and tried the most recent Lakka image (2.2.2) and still had the same issue as before. Everything would lock up if I did anything network related.

The issue seems to be related to ethtool, I removed it as a dependency of the network package and compiled Lakka. Once it was gone everything worked perfectly.

It was probably due to the udev rules in the ethtool package and the lack of ethernet port on the C0

closed 2020-08-18 by Ntemis