libretro/Lakka-LibreELEC #370: Data transfer aborts after connecting to wireless router.

prokhyon, 2018-02-03:
After flashing the newest Lakka image (Raspberry Pi version for Raspberry Pi 3 B) to an SD card the ethernet connection via UTP cable works perfectly. I can copy files under /storage folder.

After entering my WiFi password and connecting to my wireless network everything goes wrong.

When I start a SCP/SFTP process, the upload through the wireless connection gradually start to slow down, until it completely aborts. The smaller files (1-10 kbyte) are sometimes transferred (especially if I copy them file by file), but the bigger ones (1-300 Mbyte) certainly make the connection aborted.
If I start a bigger download from Lakka menu (e.g.: upgrading the image), it stucks around 14%, the download aborts and it tries to start the download process again.

After using the cable connection again (where the raspberry pi is connected to my router), the same issue happens, the problem is the same (if i recall correctly I did not disconnected from wireless). Every other device (mobile, computers etc.) have a good download/upload speed, the connection is perfect, while Lakka struggles.

If I connect my raspberry to my Ubuntu laptop via UTP cable and I give him an IP address, the ethernet connection is perfect again, I can copy bigger files again under /storage folder, but this is not a convenient way to upload new contents.

ToKe79, 2018-02-03:
Hi,
1) are you using the built in wifi or a wifi dongle?
2) when you connect to router using ethernet and connect from other computer to the raspi using the IP assigned to wlan interface, it makes sense, that the connection would be the same - you must connect to the IP assigned to the eth interface.

Please provide also some logs - see [Troubleshooting Lakka](http://www.lakka.tv/doc/Troubleshooting-Lakka/). Thanks!

ToKe79, 2018-02-03:
Wait, when I read your post again... so to reproduce:
1. you connect raspi via eth and get an IP
2. start transfer while connected to the IP from step 1
3. connect rasp to wifi
4. disconnect eth-cable
It makes sense, that the connection from step 2 to the IP from step 1 gets interrupted when making step 4.

prokhyon, 2018-02-05:
Not exactly..

1. I connect raspi to my router via eth and I get an IP
2. until this point any data transfer succeeds perfectly via cable without abortion
...
3. after a time I remove the UTP cable and I type in my wireless password to use raspi/Lakka in a more comfortable way
4. I restart my raspberry, sometimes my router
5. it does not matter which connection (eth or wifi) I use, the data transfer always aborts on biger files (> ~1M)
...
6. I connect my raspi to my router via eth cable again
7. the data always transfer slows down then aborts (on bigger files only > ~1M)
...
8. I set up a dhcp service on my Ubuntu laptop
9. I connect my raspi to my laptop via eth cable again
10. every data transfer succeeds perfectly again

The connection is not the same as you have described in your step 2.
I can restart my raspberry and my router many times and connect via eth and wifi, but the data will abort in each case one the wifi password is entered to Lakka.

Today afternoon I will try to send you the requested logs, but I'm not sure which logs are interesting for you.

closed 2019-06-26 by ToKe79