libretro/Lakka-LibreELEC #374: XU4 PPSSPP issues

kivutar, 2018-02-04:
- Which version of Lakka are you using?

OdroidXU3 after the kernel update

- What system hardware are you using?

OdroidXU4

- What did you do?

Tried a few PSP games

- What did you expect to happen?

I expected them to work as they used to work on the previous version

- What happened instead?

One crashed retroarch

The other one have glitches

kivutar, 2018-02-04:
I did, and I cleaned RA and ppsspp. The glitches are still there.

I test on a game called fl0w

Ntemis, 2018-02-05:
Then is a core issue for this game. I will get it and test.

kivutar, 2018-02-06:
I test this game on all our supported hardware. XU4 with the new kernel and driver is the only one that get this glitches.

Also, there is another game that I used to play on XU4, and this one crashes now.
I can still play it on RA for windows for the new PPSSPP.

Maybe there is something wrong in the blobs.

orbea, 2018-02-13:
Fwiw, the ppsspp libretro core is full of bugs that I don't know how to fix and it seems almost no one else wants to spend time on the core... I wouldn't be surprised if something in the libretro implementation was the cause of this, but without the hardware I can't even try to reproduce.