libretro/Lakka-LibreELEC #378: Python3 required for vulkan-loader [Tinkerboard]

ToKe79, 2018-02-08:
Ran into build error while compiling the Lakka-2.1-dev branch on a minimal system (Debian 9.3):
```
  BUILD    vulkan-loader (target)
...[redacted]...
CMake Error at /home/vudiq/projects/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm-2.1-devel/toolchain/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find PythonInterp: Found unsuitable version "2.7.13", but
  required is at least "3" (found
  /home/vudiq/projects/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm-2.1-devel/toolchain/bin/python)
Call Stack (most recent call first):
  /home/vudiq/projects/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm-2.1-devel/toolchain/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /home/vudiq/projects/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm-2.1-devel/toolchain/share/cmake-3.6/Modules/FindPythonInterp.cmake:163 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:16 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/vudiq/projects/Lakka-LibreELEC/build.Lakka-TinkerBoard.arm-2.1-devel/vulkan-loader-1.0.61.1/.armv7ve-libreelec-linux-gnueabi/CMakeFiles/CMakeOutput.log".
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1
```
Installing Debian package `python3` (with all its dependencies) on the host system solved the issue.

So python3 should either be added to the initial test, that installs other host system packages, or the Lakka-LibreELEC python package should be upgraded, or separate python3 package added as dependency of vulkan-loader.

Ntemis, 2018-02-13:
i will remove vulkan loader or you can. I see no intention from rockchip to support vulkan on rk3288 now or in the future.

Ntemis, 2018-02-21:
bye bye

closed 2018-02-21 by Ntemis