libretro/Lakka-LibreELEC #389: What is Lakka's license?

markwkidd, 2018-02-15:
I recently took a crack at creating a LICENSE doc for this repo based on the LibreELEC license and information in the libretro docs about licensing for the libretro components. I am going to set that project aside based on feedback from @twinaphex but I'm filing this issue so that in the future someone could come back around to this.

**Starting points:**
* LibreELEC license - GPLv2, acknowledging that there are other upstream licenses - https://github.com/LibreELEC/LibreELEC.tv/blob/master/README.md
* RetroArch license - GPLv3 - https://github.com/libretro/RetroArch/blob/master/COPYING
* retroarch-assets license - CC BY 4.0 - https://github.com/libretro/retroarch-assets/blob/master/COPYING
* List of non-commercial libretro emulator cores: https://buildbot.libretro.com/.docs/tech/licenses/

twinaphex, 2018-02-15:
We are not going to be pushed into removing our non-commercial libretro cores or anything, and signalling to the world we are GPL to a bunch of clueless entrepreneurs would be a disaster.

RetroPie and Recalbox have done the exact same thing for the past 7+ years, I see no reason here to give any entrepreneurs the wrong ideas. I do not want there to be even more problems in the future, Lakka in its public release form should be off-limits for any kind of commercial use. 

Do not keep pushing this issue, I do not appreciate where this is going.

closed 2018-02-15 by twinaphex

markwkidd, 2018-02-15:
I'm not opening any more PRs or Issues, but I think there will be other people who wonder: when I contribute code or other IP to Lakka, what license am I contributing under? LibreELEC has a clear answer to that question.

I'll be moving on to other areas now, though.

twinaphex, 2018-02-15:
'when I contribute code or other IP to Lakka, what license am I contributing under?' -

Lakka factually does not exist as anything other than a Linux distro that just comes with a pretty much unaltered version of RetroArch.

markwkidd, 2018-02-15:
Got it. Thank you!

ClosetMonkey, 2019-01-15:
I think this issue should be further addressed. Certain non-commercial licenses, including GPL, require license files to be easily accessible to the end-user to the point that an end-user (or redistributor if the case may be) can't reasonably acquire and then use/redistribute the product without viewing the license. This is ABSOLUTELY required. These licenses rely on an already flimsy precedent that the end-user / redistributor signals acceptance of terms just by simply using the wares or redistributing them. If it is suggested that an end-user or redistributor was denied such license information it will prove to be a big no-no for that packager, especially when public posts are made where the packager refuses to list the licences. This is just food for thought, I have no interest in giving you legal representation either.

ClosetMonkey, 2019-01-15:
Maybe someone has 1,000 Android TV boxes on order to package Lakka on. Maybe this person will make a copy of Lakka available for free download and suggest making the software freely available satisfies license concerns. Maybe the original software authors will challenge this, and Pirate Bob will link to sources where Lakka deliberately refused to share license information.

RobLoach, 2019-01-15:
> [Can I sell Lakka boxes?](http://www.lakka.tv/doc/FAQ/)
> No. Lakka is shipped with emulators protected by a Non Commercial license. Also, the name and the logo are a registered trademark.

The licenses are extremely important, and abiding by them is critical. Not only for the legalities, but also out of respect for the original authors of the emulators and cores.

You can see a list of the cores that are under a non-commercial license over at:
https://docs.libretro.com/tech/licenses/

ClosetMonkey, 2019-01-15:
You should have a "Welcome" screen after first boot that requires users to accept terms of all licenses. Every day users will just click through this which is naive and sad, but the norm. Redistributors are what your worried about however. Someone copying/redistributing the software can't reasonably say that they did so without first using it, meaning they must have previously acknowledged that they had read and accepted the terms of all licences. Beside, many of these license require that you reproduce the license text to the user in some form if used in binary form and doing it on first boot would satisfy this.

ClosetMonkey, 2019-01-15:
Let me rephrase that - copying/redistribution is generally what the authors of the included software are worried about.

natinusala, 2019-01-15:
All releases of Lakka contain a license folder with every license in it.