libretro/Lakka-LibreELEC #390: Intel GMA 3600 driver for PC

kvvinokurov, 2018-02-15:
Lakka for PC x86 32bit - http://le.builds.lakka.tv/Generic.i386/Lakka-Generic.i386-2.1.img.gz

Pegatron Amis L6 SAISHIAT (90P2-5G10030) Atom D2700
CPU: Intel Atom D2700 2.13Ghz Dual Core
Chipset: Intel NM10 (CG82NM10)
Video: Intel GMA 3600
Audio: Realtek ALC662

Stuck when loading in live mode. Here my log files: https://cloud.mail.ru/public/LWi3/7MzVsDb6M

In retroarch log found:
> pci id for fd 3: **8086:0be2**, driver (null)
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No such file or directory
failed to load driver: kms_swrast
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory
failed to load swrast driver
[WARN] [KMS]: Couldn't create GBM device.
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)
> 
8086:0be2 is Intel GMA 3600

Thread in forum: https://forums.libretro.com/t/stuck-on-boot-after-install-and-liveboot-from-usb/14777

gouchi, 2018-02-16:
As I replied on the forums, Intel GMA 3600 is based on [PowerVR](https://dri.freedesktop.org/wiki/PowerVR/) and it is not supported on Linux. Moreover, we are not using Xorg or Wayland on Generic platform.

kvvinokurov, 2018-02-18:
For ubuntu like already there is a driver. [http://linuxeeepc.blogspot.ru/2012/08/lubuntu-on-eeepc-1025c-with-correct.html](http://linuxeeepc.blogspot.ru/2012/08/lubuntu-on-eeepc-1025c-with-correct.html)

gouchi, 2018-02-18:
It won't work for kernel 4.11.12 we are using.

closed 2018-02-23 by gouchi