libretro/Lakka-LibreELEC #393: S905 Lakka dont work on AZW w95 android box with 2gb ram cvbs 480 needed

edneyhelene, 2018-02-15:
Lakka S905 port doesnt works on my box!
Tried almost everything!
it doesnt boots on,  maybe dtb.img needs to be different, i dont know!

ITs a shame because i only bought it to use s905 port !

On website i find no instructions at all on how to flash it!

However, im very used to these proceeds and i was able to flash libreelec without errors. Lakka team please do something and fix it! i need it!

480p cvbs is for sure the reason on why it's not booting, you gotta enable it on kernel or something so people like me can boot lakka on s905 devices without problem using av output

Hardware Details:

AZW W95 Android Box
S905W processor
2gb Ram
16Gb rom!
Android Nougat

ToKe79, 2018-02-16:
Did you replace the `dtb.img` with the correct one for your device from [here](https://kszaq.libreelec.tv/s905/8.2/device_trees/)?

edneyhelene, 2018-02-16:
yes sure i did, the problem is resolution....need 480cvbs mode to be inserted into its code. libreelec runs fine! retroarch and emulationstation doesnt.

its frame broken on screen....

Ntemis, 2018-02-21:
HW limitation then, closing

Ntemis, 2018-02-21:
Not sure if can be done on retroarch with this hw

ToKe79, 2018-02-21:
@edneyhelene I have a [RetroArch KODI addon](http://www.vudiq.sk/repository_kodi/repository.vudiq/repository.vudiq-1.0.0.zip) - you may try to install it from there. At least to test if RetroArch will work. The installation will take quite a long, as it is a huge archive with all the database files, assets, joypad configs, etc. - just give it some time - it will be stuck at 0% for long ;-)
The BIOS files are expected to be in ˙/storage/.config/retroarch/system` folder. See more [details about the folders here](https://github.com/ToKe79/retroarch-kodi-addon-LibreELEC#folders).

edneyhelene, 2018-03-30:
Retroarch on android plat works without problems!

The problem is that lakka itself or libretro img cant work with cvbs, i already saw similar topics and till now no news, devs needs to fix it!

i also bough an x96 box and the same problems happens! boot up libreelec perfect, but then i open retroarch or lakka and its gone...

CVBS output is not working at all and needs a fix. only works on android but i dont like the android apk....and as far as lots of projects does uses it as base, im really needing a fix for this issue with devs!

here:

https://forums.libretro.com/t/how-could-i-utilize-composite-tv-out-on-amlogic-s805/10866/2

And here: https://forums.libretro.com/t/composite-video-on-lakka-amlogic-s905x/12007

So when are you guys going to fix it?

edneyhelene, 2018-03-31:
Manual installing on libreelec does the same, no news and still no fix avaliable..

only works with hdmi output..... the android apk works fine.

ToKe79, 2018-04-04:
Hi,
probably due to the default configuration in retroarch.cfg (just shooting blanks here...).
Can you [connect via ssh](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/) to your box and post the output of the following command:
```
for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done
```
If there are multiple outputs/connectors, you may also try to edit retroarch.cfg (`nano /storage/.config/retroarch/retroarch.cfg` iirc) and change the line `video_monitor_index = "0"` to a different value (1, 2, etc. - depending on the number of outputs/connectors and their order).

And please do not get upset if something is not fixed right away - the developers do not have access to every kind of device, so very often it is hard to troubleshoot the issues without the exact same or at least similar device.

bonzini, 2019-01-01:
This is not related to the initramfs, it's a bug in retroarch and it shows up also when using the various Kodi addons that start retroarch.

bonzini, 2019-03-09:
You need to run [this script](https://raw.githubusercontent.com/shantigilbert/CoreELEC/Sx05RE/packages/sx05re/sx05re/bin/setres.sh) before starting retroarch. Then it works great. :)

edneyhelene, 2019-10-31:
> You need to run [this script](https://raw.githubusercontent.com/shantigilbert/CoreELEC/Sx05RE/packages/sx05re/sx05re/bin/setres.sh) before starting retroarch. Then it works great. :)

How do i make it?