libretro/Lakka-LibreELEC #394: Cannot mount USB HDD on Raspberry Pi

vickyorlo, 2018-02-17:
I'm running Lakka 2.1 (most recent stable at the time of writing) via NOOBS on the Raspberry Pi 3.

I connected an external USB HDD to the Pi in order to open content on it via RetroArch, but despite being reassured that any HDD, even an NTFS on should be automatically mounted in /storage/roms, it does not appear there. 
Note that both Raspbian and Kodi deal with the drive just fine.

After talking to @kivutar, it turns out that udevil seems to be the culprit here, as trying to start it brings up an error:

``Failed to start udevil-mount@.service: Unit name udevil-mount@.service is missing the instance name.
See system logs and 'systemctl status udevil-mount@.service' for details.``

Following the guidelines in [debugging udev rules](http://www.jpichon.net/blog/2011/12/debugging-udev-rules/), ``udevadm test /dev/sda1`` returns this error:
``Invalid rule /usr/lib/udev/rules.d/73-seat-late.rules:15: RUN{builtin}: 'uaccess' unknown``

A cursory Google search brings up a person with a similar issue who resolved it by simply deleting the file. It does not seem to be much of a solution, though, not to mention the filesystem is read only in Lakka.

For now I've simply set up a RetroPie on my Pi as a workaround and it seems to work fine.

gouchi, 2018-03-17:
Is it possible to get the output of ?
`systemctl status udevil-mount@.service`

Do you reproduce the issue with [nightly build](http://nightly.builds.lakka.tv/canary/) ?

kivutar, 2018-03-19:
I just tried with a FAT32 USB drive on my RPi3 with the latest nightly build and the feature works as expected.

vickyorlo, 2018-03-30:
@gouchi: No, it is not. Instead I get the aforementioned "Unit name udevil-mount@.service is missing the instance name. " error. Nightly doesn't seem to do much, either.

ToKe79, 2018-06-07:
@vickyorlo does it also happen when you boot directly from SD card or only via NOOBS? You can get latest development version [here](http://nightly.builds.lakka.tv/2018-05-28/).

ToKe79, 2018-07-02:
Please test the service by:
```
systemctl status udevil-mount@-dev-sdxn
```
where `-dev-sdXn` should be replace by the device (`/dev/sdb1` => `-dev-sdb1`, `/dev/sda2` => `-dev-sda2`, etc.) - e.g. `systemctl status udevil-mount@-dev-sda1` in my test case gives output like this (after plugging in USB - /dev/sda with 2 partitions - with my Lakka test image):
```
udevil-mount@-dev-sda1.service - Udevil mount service
   Loaded: loaded (/usr/lib/systemd/system/udevil-mount@.service; static; vendor preset: disabled)
   Active: active (exited) since Mon 2018-07-02 21:29:54 CEST; 5min ago
  Process: 4348 ExecStartPost=/usr/lib/samba/samba-autoshare (code=exited, status=0/SUCCESS)
  Process: 4336 ExecStart=/usr/bin/udevil --mount %I (code=exited, status=0/SUCCESS)
 Main PID: 4336 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/system-udevil\x2dmount.slice/udevil-mount@-dev-sda1.service

Jul 02 21:29:54 Lakka systemd[1]: Starting Udevil mount service...
Jul 02 21:29:54 Lakka udevil[4336]: Mounted /dev/sda1 at /storage/roms/LAKKA
Jul 02 21:29:54 Lakka systemd[1]: Started Udevil mount service.
```
Unfortunately I do not have any NTFS disk by hand, so cannot test NTFS auto-mount.

ToKe79, 2019-07-02:
no response

closed 2019-07-02 by ToKe79