libretro/Lakka-LibreELEC #396: Support Request: Compatibility with the VIA nano U3500  CPU

fange26, 2018-02-18:
Unable to run lakkatv due to CPU compatibility. 

http://www.cpu-world.com/CPUs/Nano/VIA-Nano%20U3500.html

gouchi, 2018-02-18:
Hi,

[CONFIG_CPU_SUP_CENTAUR](https://github.com/torvalds/linux/blob/master/arch/x86/Kconfig.cpu#L442) is enabled for [x86_64](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/projects/Generic/linux/linux.x86_64.conf#L442) and [i386](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/projects/Generic/linux/linux.i386.conf#L424) which should work for [Via Nano](https://en.wikipedia.org/wiki/VIA_Nano) as it has been designed by [Centaur Technology](https://en.wikipedia.org/wiki/Centaur_Technology).

So you can't boot at all ? If you can boot please boot with `live textmode` in the boot entry and try to provide `dmesg` output and [graphic log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#graphic-card-logs).

If you can't boot, did you try with another linux distribution so that we can get the output 

`cat /proc/cpuinfo`
`lspci -nnk | grep -A 3 VGA`

Thank you.

fange26, 2018-02-19:
[image]
[image]
[image]
[image]

gouchi, 2018-02-19:
Hi,

As I was afraid it is using VIA chipset which is not in a good shape on Linux but the [Openchrome team](https://lists.freedesktop.org/archives/openchrome-devel/) is making progress. As far as I know they are working on [DRM/KMS and X.org](https://cgit.freedesktop.org/openchrome/) then they will work on Mesa opengl driver. On Generic platform we are using DRM/KMS.

I am sorry but you will have another device to use Lakka.

closed 2018-02-23 by gouchi