libretro/Lakka-LibreELEC #408: Intel Compute Stick (STCK1A8LFC) wifi and bluetooth not working.

IvanPozdnyakov, 2018-03-02:
I built from https://github.com/libretro/Lakka-LibreELEC/commit/0249d6d5673c6f5e90d2d182504bc5a6b90a248c and installed on an Intel STCK1A8LFC.

The Intel Compute Stick uses a realtek chip that has both Bluetooth and Wifi on it. From doing some research I couldn't find any official link to a realtek driver but apparently you can reach out and they will provide it. I *believe* that **rtl8723bs** is the appropriate driver and I was able to find it in the repo below:

https://github.com/hadess/rtl8723bs

Some helpful resources suggesting that similar issues occur on other distros, so far I only tested by compute stick with ubuntu 14.04 and it worked without issue:

https://linuxiumcomau.blogspot.com/2016/04/ubuntu-1604-iso-for-intel-atom-compute.html
https://liliputing.com/2015/07/ubuntu-on-the-32gb-intel-compute-stick-you-have-to-install-it-yourself.html

IvanPozdnyakov, 2018-03-02:
Additional asks for this can be found on the forum:

https://forums.libretro.com/t/intel-compute-stick-wifi/6575/4
https://forums.libretro.com/t/intel-compute-stick/2798/3

IvanPozdnyakov, 2018-03-03:
Awesome, thanks! Pulled your branch and building now. Will confirm when tested (may take a while, only my NUC has Ubuntu 16.04 installed, having issues building Lakka on any of my other developer boxes).

IvanPozdnyakov, 2018-03-03:
Is there a signature and checksum available for images you provided? This is actually a personal wish-list item for the official release page on the site as well! I wasn't able to find those anywhere. I'm not sure if this item is appropriate as a Github issue.

ToKe79, 2018-03-03:
Hi,
I added the checksum files in the folder.
Also for releases the checksums are in the same folder (md5, sha256) as the images ([le.builds.lakka.tv](http://le.builds.lakka.tv)), but the download page ([lakka.tv/get](http://lakka.tv/get)) does not point to them - usually users even do not know how to check those.

IvanPozdnyakov, 2018-03-05:
Tried it, but you were right it wasn't enough, kind of unfortunate but I guess it's on Intel for picking the unusual hardware that they did for their compute stick line. This is probably an edge case and doesn't occur with most regular x86 machines.

Anyways, I *think* everything else is functioning properly and I don't think I have a requirement for connecting the device to my network. The USB drivers seem to work since the input from my DS3 controller is recognized. I'll deal with the slight inconvenience of having to use a wire! :smiley: 

I can close the issue, or leave it if someone still wants to look into it further?

gouchi, 2018-03-10:
The wifi driver will be in staging area for [kernel 4.12](https://kernelnewbies.org/Linux_4.12#Drivers_in_the_Staging_area).
For the bluetooth, we will need [rtl8723bs_bt](https://github.com/lwfinger/rtl8723bs_bt).

gpapp, 2018-10-15:
This issue also affects low budget chineese tablets (Teclast X80 Pro) and some intel based cheap TV boxes, that would be a natural target for Lakka.

gouchi, 2018-10-23:
For the wifi, it should be supported when #556 is fixed.

zezba9000, 2018-12-03:
Running into this issue as well.

Shanzie82, 2019-02-15:
Hey guys. I've had this problem every since I purchased it about a year ago. I put it in a box in the closet and forgot about it. I was cleaning the other day and found. I found this link ( https://itsfoss.com/fix-no-wireless-network-ubuntu/ ) this morning. I'm gonna try it out and see if it helps. I hope it could help you. I'll let you know later.

gouchi, 2019-03-08:
@CanntAim can you try this [testing image](https://drive.google.com/drive/folders/128MaH09bjVcB2pR4Kj2XxdBabR-oDP75) ?

Thank you.

CanntAim, 2019-03-28:
Sure @gouchi, I'll try and get to it sometime in the next couple of weeks probably, a little bit busy now with other stuff.

Thanks for the support on this!

gouchi, 2019-07-16:
@CanntAim We have now [nightly builds based on LE 9.2](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/) if you want to test.

Shanzie82, 2019-07-16:
Hello everyone, not sure if it will work for everyone but I was downloaded PDAnet+ from the playstore and was able to get internet from my cellphone via USB connection. I purchased a 5 port hub w/on off switch. Plugged the plugged the phone, keyboard&mouse and whatever else. Enabled the app, followed the steps in the app. Switched the connection on the  compute stick and was connected. Hope this helps. Now if I can figure how to get more RAM by using a flashdrive...

closed 2020-08-18 by Ntemis

CanntAim, 2020-08-19:
Ah, sorry, I may have not gotten the chance to test this last year like I said I would. Does this still need testing? I still may have the stick lying around somewhere but I need to find it and my time availability is a bit limited right now.

Pretty sure Intel still supports this model with BIOS updates so it's not exactly deprecated and maybe worth fixing but I'm not sure if this bears any relevance to their newer models.

Ntemis, 2020-08-20:
Feel welcome to test and report any issues by testing the newer nightly image you can find here https://nightly.builds.lakka.tv/