libretro/Lakka-LibreELEC #409: Strange behavior with xbox360 controller

martinotu94, 2018-03-11:
Which version of Lakka are you using? 2.1 
What system hardware are you using? RPi zero W
What did you do? Try to use xbox 360 controller to play and move in menus
What did you expect to happen? good movement with the controller
What happened instead? after some random time the controller start to stuck in one direction and after a couple o second stop completely to response to input. After unplugging it and replugging everything begin to work again for some time ad after that the same problem show up. Quite frustrating to use :(

All this while trying to write the wifi password witch for another bug was impossible to enter with the keyboard.

RobLoach, 2018-03-11:
Analog to Digital controller enabled?

martinotu94, 2018-03-11:
Sorry but i don't know what you are talking about. Can you explain?
thank you a lot

RobLoach, 2018-03-11:
[image]

martinotu94, 2018-03-11:
I'll try asap and let you know!

martinotu94, 2018-03-17:
Even enabling analog to digit i keep having the same problem. After some time the input stop responding. 

Trying on the latest available build

ToKe79, 2018-06-05:
@martinoturrina can you try latest nightly build, e.g. from here: http://nightly.builds.lakka.tv/2018-05-28/
Should at least fix the keyboard issue.

martinoturrina, 2018-06-05:
@ToKe79 yes, the latest nightly build have this problem fixed! hope to see this fix on final release soon :)

ToKe79, 2018-06-05:
@martinoturrina Is also the controller issue fixed?

martinoturrina, 2018-06-05:
@ToKe79 no, sadly the issue is already there!

closed 2020-08-18 by Ntemis