libretro/Lakka-LibreELEC #411: Shutdown with Wetek_Core

chataigne73, 2018-03-13:
Hello
LAKKA works well on my wetek core, but there is one big: when selecting shutdown in Lakka menu, the wetek core reboots instead of powering off.
Thank you

chataigne73, 2018-03-16:
Is anybody still here???

gouchi, 2018-03-17:
Which version are you using ? Do you reproduce the issue with [nightly builds](http://nightly.builds.lakka.tv/canary/) ?

Thank you.

chataigne73, 2018-03-18:
I'm using last 2.1 version.
I will test nightly.
Just one more information: shutdown is working ok in libreelec

kivutar, 2018-03-19:
> Just one more information: shutdown is working ok in libreelec

LibreELEC 8.2 ?

chataigne73, 2018-03-19:
yes, LibreELEC 8.2

Ntemis, 2018-06-02:
@codesnake any thoughts?

codesnake, 2018-06-05:
@Ntemis WeTek Core and WeTek Play uboot doesn't support poweroff, only suspend, so if you run poweroff it just reboots the box. As a workaround LE uses a patch for Kodi: https://github.com/LibreELEC/LibreELEC.tv/blob/libreelec-8.2/projects/WeTek_Core/patches/kodi/0004-powermanagement-Perform-suspend-instead-of-powerdown.patch