libretro/Lakka-LibreELEC #413: Scanning files is broken

MrUnny, 2018-03-17:
When i go to add a new game to my collection, it does not add the game, say if i want to add a new SNES game, it will not add, i tried rebooting, that still did not work, even removing the game and trying to add another one will not work.

RobLoach, 2018-03-17:
Mind providing the CRC of the file? Name of the game too?

MrUnny, 2018-03-17:
So when I added Contra 3 which was a .smc file i wanted to add Kirby's Dream Course, also a .smc file, i would scan the file, and it would not work, i even tried adding Super Bomberman to make sure it was not the rom, sure enough, it still did not work

RobLoach, 2018-03-17:
Mind sharing the CRC of the files?

MrUnny, 2018-03-17:
I have no idea how to get the CRC

kivutar, 2018-03-17:
I'm getting this problem too on some systems. Take a look at Neo Geo Pocket roms @RobLoach none are scanning for me with the latest database.
No-Intro must have updated them, and the roms that people used to use are not compatible anymore.

RobLoach, 2018-03-17:
Contra III for SNES...
http://datomatic.no-intro.org/index.php?page=show_record&s=49&n=3612
FFDF4355

Scanned okay here... Depending on your platform, you get get the crc32 of a file https://duckduckgo.com/?q=get+crc32+of+file&ia=web . I use `crc32 FILENAME` (Linux).

kivutar, 2018-03-18:
All roms are scanning OK in my tests for https://github.com/libretro/Lakka-LibreELEC/commit/24fea9f17b360f31aeee6d6a686fd8d704dc5d8c

closed 2018-03-18 by kivutar