libretro/Lakka-LibreELEC #415: Amiga roms wont load

jdca, 2018-03-17:
Lakka-RPi2.arm-2.1-devel-20180305031024-r28154-gb552a9a26.img.gz

Amiga roms doesnt load att all.
Got 192d6d950d0ed3df8040b788502831c2  kick13.rom in system folder
This is also in 2.1 stable.

gouchi, 2018-03-17:
Please provide some log especially [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

jdca, 2018-03-17:
Lakka:~/roms # systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.7.0 (Git 3a981e081c)
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Mar  5 2018
[INFO] Version: 1.7.0
[INFO] Git: 3a981e081c
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: videocore
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[... 208 lines cut ...]
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [autoconf]: selected configuration: /tmp/joypads/udev/8Bitdo_Pro_N30_BT.cfg
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x2804fc
 (0)=1 
 (1)=1 
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] Written to playlist file: /storage/.config/retroarch/content_history.lpl
Uae4arm v0.4 for Raspberry Pi by Chips
starting sound thread..
retro:(320,240) gfx(640,256)
starting sound thread..
snd
starting sound thread..
snd

gouchi, 2018-03-24:
I tried on Asus Tinker Board and it is working with puae and uae4arm but I used [uae cfg](https://github.com/libretro/libretro-uae/blob/master/RickDangerous.uae) file to launch the game.

jdca, 2018-03-25:
Ah yes you were right, my paths was wrong in my cfg's

P-UAE works but not UAE4ARM.
192d6d950d0ed3df8040b788502831c2  kick13.rom
Lakka-RPi2.arm-2.1-devel-20180322014750-r28164-gc90e96b2e

[INFO] Updating firmware status for: /tmp/cores/uae4arm_libretro.so on /storage/system
[INFO] Using content: /storage/roms/FreeNAS/Favoriter/Commodore/Amiga/Another World.adf.
[INFO] arg #0: retroarch
[INFO] arg #1: /storage/roms/FreeNAS/Favoriter/Commodore/Amiga/Another World.adf
[INFO] arg #2: -s
[INFO] arg #3: /storage/savefiles
[INFO] arg #4: -S
[INFO] arg #5: /storage/savestates
[INFO] arg #6: -c
[INFO] arg #7: /storage/.config/retroarch/retroarch.cfg
[INFO] arg #8: -L
[INFO] arg #9: /tmp/cores/uae4arm_libretro.so
[INFO] Threaded video stats: Frames pushed: 979, Frames dropped: 0.
[INFO] Set config file to : /storage/.config/retroarch/retroarch.cfg
[INFO] RetroArch 1.7.1 (Git 29a057ac98)
[INFO] Redirecting save file to "/storage/savefiles/Another World.srm".
[INFO] Redirecting savestate to "/storage/savestates/Another World.state".
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Mar 22 2018
[INFO] Version: 1.7.1
[INFO] Git: 29a057ac98
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/uae4arm_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/uae4arm/uae4arm.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/uae4arm/Another World.cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/uae4arm/Another World.cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/uae4arm/Another World.glslp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/uae4arm/Amiga.cgp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/uae4arm/Amiga.glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/uae4arm/uae4arm.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/uae4arm/uae4arm.glslp.
[INFO] Environ SET_CONTROLLER_INFO.
[INFO] Controller port: 1
[INFO]    AMIGA Joystick (ID: 513)
[INFO]    AMIGA Keyboard (ID: 259)
[INFO] Controller port: 2
[INFO]    AMIGA Joystick (ID: 513)
[INFO]    AMIGA Keyboard (ID: 259)
[INFO] Environ SET_VARIABLES.
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/uae4arm/Another World.rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/uae4arm/uae4arm.rmp.
[INFO] Redirecting save file to "/storage/savefiles/Another World.srm".
[INFO] Redirecting savestate to "/storage/savestates/Another World.state".
[INFO] Environ SYSTEM_DIRECTORY: "/storage/system".
[INFO] Environ CORE_ASSETS_DIRECTORY: "/storage/roms/downloads".
Retro SYSTEM_DIRECTORY /storage/system
Retro SAVE_DIRECTORY /storage/savefiles
Retro CONTENT_DIRECTORY /storage/roms/downloads
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Environ SET_INPUT_DESCRIPTORS:
[INFO] 	RetroPad, User 1, Button "B (bottom)" => "B"
[INFO] 	RetroPad, User 1, Button "Y (left)" => "Y"
[INFO] 	RetroPad, User 1, Button "Select" => "Select"
[INFO] 	RetroPad, User 1, Button "Start" => "Start"
[INFO] 	RetroPad, User 1, Button "D-Pad Up" => "Up"
[INFO] 	RetroPad, User 1, Button "D-Pad Down" => "Down"
[INFO] 	RetroPad, User 1, Button "D-Pad Left" => "Left"
[INFO] 	RetroPad, User 1, Button "D-Pad Right" => "Right"
[INFO] 	RetroPad, User 1, Button "A (right)" => "A"
[INFO] 	RetroPad, User 1, Button "X (up)" => "X"
[INFO] 	RetroPad, User 1, Button "L" => "L"
[... 112 lines cut ...]
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x2984f0
 (0)=1 
 (1)=1 
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
Uae4arm v0.4 for Raspberry Pi by Chips
starting sound thread..
retro:(320,240) gfx(640,256)
starting sound thread..
snd
starting sound thread..
snd

gouchi, 2018-03-25:
Can you try launch with the cfg file directly which will load the adf file ?

jdca, 2018-03-25:
Yeah that's what I'm doing. 

My Another World.adf looks like this

floppy0=/storage/roms/FreeNAS/Favoriter/Commodore/Amiga/amigaroms/anworld1.adf
floppy1=/storage/roms/FreeNAS/Favoriter/Commodore/Amiga/amigaroms/anworld2.adf
kickstart_rom_file=/storage/system/kick13.rom

cpu_speed=real
chipset=aga
cpu_type=68000
compfpu=false
cpu_compatible=false
cachesize=0

chipmem_size=4
fastmem_size=8
z3mem_size=128
bogomem_size=0
gfxcard_size=16
scsi=false

rtg_nocustom=true
rtg_modes=0x212

sound_output=normal
sound_frequency=20050
sound_channels=stereo

show_leds=false
show_leds_rtg=true

config_description=
config_hardware=false
config_host=true

gfx_display=0
gfx_framerate=1
gfx_width_windowed=800
gfx_height_windowed=600
gfx_width_fullscreen=800
gfx_height_fullscreen=600
gfx_refreshrate=0
gfx_vsync=false
gfx_lores=false
gfx_linemode=double
gfx_fullscreen_amiga=true
gfx_fullscreen_picasso=true
gfx_center_horizontal=simple
gfx_center_vertical=simple
gfx_colour_mode=16bit

gouchi, 2018-03-25:
Can you try with a [simple one](https://github.com/libretro/libretro-uae/blob/master/RickDangerous.uae) then add the other parameters ?

Thank you.

jdca, 2018-03-25:
It kinda worked, couldnt make the sound work or the resolution but it started.

Gonna try some more tomorrow after work. 

Thanks.

jdca, 2018-03-27:
So I got this to work with some tinkering with the config file pretty good with my raspberry pi 3 and P-UAE 2.6.1
Git Commit: 0186c1b16f7181ffa02d73e6920d3180ce457c46
Build date: Mar  5 2018 04:27:02

and

Lakka-RPi2.arm-2.1-devel-20180305031024-r28154-gb552a9a26

I use this "Nitro.txt" renamed too Nitro.uae to play the game pretty flawless without any studdering.
Got some issue with the cpu but solved that with 

cpu_compatible=false
cpu_cycle_exact=false

Now my cpu doesnt go over 75%

And some issue with P-UAE and my 8bitdo n30 pro acting as a mouse when my analog sticks was enabled, when I removed the inputs on the analog sticks it solved that issue but thats not very smooth.

I then added 

joyport0=joy0
joyport1=mouse

And when entering the game i pressed select once (switching from mouse to joy) everything works.

If not, letters when choosing name on my character goes highwire and I can not steer my car as it turns every direction all the time :/

When starting P-UAE with systemctl stop retroarch.service ; retroarch -v
All i could see was "inputdevice change 'DEFMOUSE1':0->0"

No idea if that has something to do with anything.

Im also playing on an old tv so the gfx lines maybe dont work so good with new tv's

[Nitro.txt](https://github.com/libretro/Lakka-LibreELEC/files/1852794/Nitro.txt)

gouchi, 2018-03-31:
Thank you for the feedback.

I am little confused it is working or not at the end ?

jdca, 2018-03-31:
P-UAE on RPi3 works with the game Nitro
P-UAE on my PC x86/AMD64 do not work with the game Nitro I get "Segmentation fault (core dumped) as mentioned.

And i can't, for example get Leisure Suit Larry in the Land of the Lounge Lizards to start because some buttons in P-UAE seems to be hardcoded and i need to answer some questions before i can actually play the game, so for example when i press button A the P-UAE menu shows up instead of answering the question.

Start shows me a virtual keyboard
Select switches from joystick to mouse or vice versa
L2 Shows me some kinda mouse/joy config
R2 Toggles the sound
Select + A emulates left mouse button?
Some games needs "Enter" to continue loading and then i get the virtual keyboard in my face

I guess i could use the virtual keyboard to answer the question but if i could map my own keys that would be an improvement.

So for the game Nitro, yes it works.
Other games the answer is so so.

With UAE4ARM Nitro runs way to fast and the resolution is way off but the game starts.
And if i use lines from my P-UAE config on my UAE4ARM

systemctl stop retroarch.service ; retroarch -v

tells me this.

(I added 4 lines)

gfx_center_vertical=simple
gfx_center_horizontal=simple
gfx_width=640
gfx_height=420

And this is the error.

Uae4arm v0.4 for Raspberry Pi by Chips
starting sound thread..
retro:(320,240) gfx(640,420)
update_display
*** Error in `retroarch': munmap_chunk(): invalid pointer: 0x710a9868 ***
Aborted (core dumped)

gouchi, 2018-03-31:
So I can close this issue as it is working on RPi with puae with uae configuration ?

jdca, 2018-03-31:
Yes, the roms loads.

closed 2018-03-31 by gouchi