libretro/Lakka-LibreELEC #417: 8bitdo nes30 pro gamepad

jdca, 2018-03-17:
Lakka-RPi2.arm-2.1-devel-20180305031024-r28154-gb552a9a26

When connecting it, it autoscrolls until you hit a button, then it stops and everything is fine.

This is also in 2.1 stable.

gouchi, 2018-03-24:
Can you try to make a test by [resetting the joypad configuration file and rebind it](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6) ?

Thank you.

jdca, 2018-03-25:
Okey, so I used Lakka-RPi2.arm-2.1-devel-20180322014750-r28164-gc90e96b2e this time.

It did the weird autoscrolling thing as usually, then i ssh'd in and moved all files under /tmp/joypads/udev to another place and connected again and that worked fine, no autoscrolling.

When you set the buttons under input its very sensitive, it's easy to set 2 buttons on the same click
Not a big deal, just anyoing sometimes.