libretro/Lakka-LibreELEC #422: RPi 3B+ failing to boot with latest release

ldilley, 2018-03-18:
Which version of Lakka are you using?
Lakka-RPi2.arm-2.1 and also attempted to use Lakka-RPi2.arm-2.1-devel-canary-d4d40fd and Lakka-RPi2.arm-2.1-devel-20180305031024-r28154-gb552a9a26.img (nightly development releases) when the first release failed.

What system hardware are you using?
Raspberry Pi 3B+ with a SanDisk Ultra 32GB class 10 microSD card and 5.25V 2.4A power supply

What did you do?
Used the latest release of Etcher for Windows 10 to flash the above un-gzipped images of Lakka onto the microSD card referenced in the hardware answer above.

What did you expect to happen?
I expected to see the Lakka splash screen.

What happened instead?
No HDMI output, no continuous red LED -- 4 long green LED blinks, followed by 4 quick green LED blinks which repeats in that pattern (4 slow followed by 4 fast green.)

Other noteworthy stuff:
The latest release of Raspbian (2018-03-13-raspbian-stretch) and NOOBS 2.7 boot normally after flashing the same SD card and using said card in the same RPi.

gouchi, 2018-03-18:
Hi,

When this [PR](https://github.com/libretro/Lakka-LibreELEC/pull/419) is merged, we will support RPi 3B +.

ldilley, 2018-03-19:
Excellent! Thank you for acknowledging, @gouchi.

closed 2018-03-19 by ldilley

SapphireTactician, 2018-03-25:
Just bought a 3B+ myself. Noticed the PR #419 was merged but then it looks like commit 
f2bf715 reverts that...? Flashing and running the Nightly dated from the 22nd http://nightly.builds.lakka.tv/2018-03-22/RPi2.arm/ still results in the same error described above, perhaps because it was reverted. Sadly I don't see a nightly from while it was merged meaning I'm stuck with a brick. :(  

ETA on when this will be fixed (*again*) so I can finally relax and enjoy some games?

alancnet, 2018-05-13:
@SapphireTactician @gouchi  I believe this issue has regressed.

I have to download the linked build above to get it to work. If I update from the menu, or download a release build from the website, I get green blinks from the board.

4 slow blinks followed by 4 fast blinks.

SapphireTactician, 2018-05-13:
@alancnet Both methods you mentioned pull from the same source. The stable release is very old. You'll need to use a nightly or compile from source until further notice.

NearistheEnd, 2018-05-19:
Thanks for the nightly was able to install. Will it break if I update?

ToKe79, 2018-05-19:
If everything runs as should, do not update via Lakka /RA menu - this downloads images from le.builds.lakka.tv (releases, such as stable, RCs etc.). So in fact it will downgrade and these older version do not support RPi3B+.

NearistheEnd, 2018-05-19:
Thanks. I wanted to update to get the customization options for the menus this build is missing them. I will just have to wait until the full build is released that supports b plus.

SapphireTactician, 2018-05-20:
>Will it break if I update?

@NearistheEnd Yes, as ToKe79 said, avoid using the updater via RetroArch as it pulls from Stable, which is the same as the website's image. Both are very out of date. Stick to nightlies and compiling from source.

floowsnaake, 2018-10-24:
im having problems