libretro/Lakka-LibreELEC #424: Atari - 5200 PC

jdca, 2018-03-20:
It scans fine but when i try to start it thru the playlist it doesnt "autostart" 
I have to choose a core manual "Atari 8-bit computer systems and 5200 (Atari800)" shows up.

When i choose that core i get an image saying Emulation halted and that it needs a real atari/os.

I couldnt find any information about any bios needed, i tried with this in my system folder.
281f20ea4320404ec820fb7ec0693b38  5200 BIOS (U).rom
and 
281f20ea4320404ec820fb7ec0693b38  5200.rom
without any luck.

If there allready is a thread about this just delete this one.

[INFO] Using content: /storage/roms/FreeNAS/Roms/Atari/Atari-5200/Beamrider (USA).zip#Beamrider (USA).a52.
[INFO] arg #0: retroarch
[INFO] arg #1: /storage/roms/FreeNAS/Roms/Atari/Atari-5200/Beamrider (USA).zip#Beamrider (USA).a52
[INFO] arg #2: -s
[INFO] arg #3: /storage/savefiles
[INFO] arg #4: -S
[INFO] arg #5: /storage/savestates
[INFO] arg #6: -c
[INFO] arg #7: /storage/.config/retroarch/retroarch.cfg
[INFO] arg #8: -L
[INFO] arg #9: /tmp/cores/atari800_libretro.so
[INFO] Threaded video stats: Frames pushed: 187, Frames dropped: 0.
[INFO] Set config file to : /storage/.config/retroarch/retroarch.cfg
[INFO] RetroArch 1.7.0 (Git 3a981e081c)
[INFO] Redirecting save file to "/storage/savefiles/Beamrider (USA).srm".
[INFO] Redirecting savestate to "/storage/savestates/Beamrider (USA).state".
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES 
Built: Mar  4 2018
[INFO] Version: 1.7.0
[INFO] Git: 3a981e081c
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/atari800_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/Atari800/Atari800.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/Atari800/Beamrider (USA).cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Atari800/Beamrider (USA).cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Atari800/Beamrider (USA).glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Atari800/Atari800.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Atari800/Atari800.glslp.
[INFO] Environ SET_CONTROLLER_INFO.
[INFO] Controller port: 1
[INFO]    ATARI Joystick (ID: 513)
[INFO]    ATARI Keyboard (ID: 259)
[INFO] Controller port: 2
[INFO]    ATARI Joystick (ID: 513)
[INFO]    ATARI Keyboard (ID: 259)
[INFO] Environ SET_VARIABLES.
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/Atari800/Beamrider (USA).rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/Atari800/Atari800.rmp.
[INFO] Redirecting save file to "/storage/savefiles/Beamrider (USA).srm".
[INFO] Redirecting savestate to "/storage/savestates/Beamrider (USA).state".
[INFO] Environ GET_LOG_INTERFACE.
[INFO] Environ SYSTEM_DIRECTORY: "/storage/system".
[INFO] Environ CORE_ASSETS_DIRECTORY: "/storage/roms/downloads".
Retro SYSTEM_DIRECTORY /storage/system
Retro SAVE_DIRECTORY /storage/savefiles
Retro CONTENT_DIRECTORY /storage/roms/downloads
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Environ SET_INPUT_DESCRIPTORS:
[INFO] 	RetroPad, User 1, Button "B (bottom)" => "B"
[INFO] 	RetroPad, User 1, Button "Y (left)" => "Y"
[INFO] 	RetroPad, User 1, Button "Select" => "Select"
[INFO] 	RetroPad, User 1, Button "Start" => "Start"
[INFO] 	RetroPad, User 1, Button "D-Pad Up" => "Up"
[INFO] 	RetroPad, User 1, Button "D-Pad Down" => "Down"
[... 323 lines cut ...]
[INFO] [GL]: Using GL_RGB565 for texture uploads.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event3).
[INFO] [udev]: Keyboard #1 (/dev/input/mouse0).
[ERROR] [udev] Failed to open device: /dev/input/event3 (Success).
[INFO] [udev]: Mouse #0 (/dev/input/mouse0).
[INFO] [udev]: Plugged pad: Wireless Controller (2578:1) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event12) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event12) supports 16 force feedback effects.
[INFO] [Autoconf]: 203 profiles found.
[INFO] [autoconf]: selected configuration: /tmp/joypads/udev/Sony-PlayStation4-DualShock4-Controller-Bluetooth.cfg
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.1-devel/alsa-lib-1.1.4.1/src/pcm/pcm_dmix.c:1099:(snd_pcm_dmix_open) unable to open slave
[ERROR] ALSA: Failed to initialize...
[ERROR] Failed to initialize audio driver. Will continue without audio.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x70e3c0
[INFO] Game focus is: : off.
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[libretro INFO] ChangeAV: w:336 h:240 ra:1.333333.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).

kivutar, 2018-03-21:
https://github.com/libretro/Lakka-LibreELEC/commit/2f415cb1b2cd23dfeeafcc040809915451487333

jdca, 2018-04-03:
Lakka-RPi2.arm-2.1-devel-20180403002029-r28169-gbbabe7b3c
Raspberry pi 3.
Roms from no-intro.
Scans fine but doesnt start.

[image]

[INFO] Using content: /storage/roms/FreeNAS/scans-ok/Atari/Atari-5200/Activision Decathlon, The (USA).zip#Activision Decathlon, The (USA).a52.
[INFO] arg #0: retroarch
[INFO] arg #1: /storage/roms/FreeNAS/scans-ok/Atari/Atari-5200/Activision Decathlon, The (USA).zip#Activision Decathlon, The (USA).a52
[INFO] arg #2: -s
[INFO] arg #3: /storage/savefiles
[INFO] arg #4: -S
[INFO] arg #5: /storage/savestates
[INFO] arg #6: -c
[INFO] arg #7: /storage/.config/retroarch/retroarch.cfg
[INFO] arg #8: -L
[INFO] arg #9: /tmp/cores/atari800_libretro.so
[INFO] Threaded video stats: Frames pushed: 315, Frames dropped: 0.
[INFO] Set config file to : /storage/.config/retroarch/retroarch.cfg
[INFO] RetroArch 1.7.1 (Git 29a057ac98)
[INFO] Redirecting save file to "/storage/savefiles/Activision Decathlon, The (USA).srm".
[INFO] Redirecting savestate to "/storage/savestates/Activision Decathlon, The (USA).state".
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Apr  3 2018
[INFO] Version: 1.7.1
[INFO] Git: 29a057ac98
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/atari800_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/Atari800/Atari800.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/Atari800/Activision Decathlon, The (USA).cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Atari800/Activision Decathlon, The (USA).cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Atari800/Activision Decathlon, The (USA).glslp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/Atari800/Atari-5200.cgp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/Atari800/Atari-5200.glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Atari800/Atari800.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Atari800/Atari800.glslp.
[INFO] Environ SET_CONTROLLER_INFO.
[INFO] Controller port: 1
[INFO]    ATARI Joystick (ID: 513)
[INFO]    ATARI Keyboard (ID: 259)
[INFO] Controller port: 2
[INFO]    ATARI Joystick (ID: 513)
[INFO]    ATARI Keyboard (ID: 259)
[INFO] Environ SET_VARIABLES.
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/Atari800/Activision Decathlon, The (USA).rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/Atari800/Atari800.rmp.
[INFO] Redirecting save file to "/storage/savefiles/Activision Decathlon, The (USA).srm".
[INFO] Redirecting savestate to "/storage/savestates/Activision Decathlon, The (USA).state".
[INFO] Environ GET_LOG_INTERFACE.
[INFO] Environ SYSTEM_DIRECTORY: "/storage/roms/FreeNAS/lakka/system".
[INFO] Environ CORE_ASSETS_DIRECTORY: "/storage/roms/downloads".
Retro SYSTEM_DIRECTORY /storage/roms/FreeNAS/lakka/system
Retro SAVE_DIRECTORY /storage/savefiles
Retro CONTENT_DIRECTORY /storage/roms/downloads
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Environ SET_INPUT_DESCRIPTORS:
[INFO] 	RetroPad, User 1, Button "B (bottom)" => "B"
[INFO] 	RetroPad, User 1, Button "Y (left)" => "Y"
[INFO] 	RetroPad, User 1, Button "Select" => "Select"
[INFO] 	RetroPad, User 1, Button "Start" => "Start"
[INFO] 	RetroPad, User 1, Button "D-Pad Up" => "Up"
[INFO] 	RetroPad, User 1, Button "D-Pad Down" => "Down"
[INFO] 	RetroPad, User 1, Button "D-Pad Left" => "Left"
[INFO] 	RetroPad, User 1, Button "D-Pad Right" => "Right"
[INFO] 	RetroPad, User 1, Button "A (right)" => "A"
[INFO] 	RetroPad, User 1, Button "X (up)" => "X"
[... 271 lines cut ...]
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event0).
[INFO] [udev]: Plugged pad: 8Bitdo NES30 Pro (11720:14368) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event2) supports 0 force feedback effects.
[INFO] [Autoconf]: 0 profiles found.
[INFO] [Autoconf]: no profiles found for 8Bitdo NES30 Pro (11720/14368).
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x2984f0
[INFO] Game focus is: : off.
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[libretro INFO] ChangeAV: w:336 h:240 ra:1.333333.

jdca, 2018-04-03:
If i create an .atari800.cfg under /storage that looks like this
with these 4 bios files (md5sums for those who want to try)
eb1f32f5d9f382db1bbfb8d7f9cb343a  atariosa.rom
a3e8d617c95d08031fe1b20d541434b2  atariosb.rom
06daac977823773a3eea3422fd26a703  atarixl.rom
281f20ea4320404ec820fb7ec0693b38  5200.rom

I end up in Atari computer - memo pad.
[image]

I then press F1, the run atari program.
Choose my rom and it works.

This is my config

Atari 800 Emulator, Version 3.1.0
ROM_OS_A_NTSC=
ROM_OS_A_PAL=/storage/roms/FreeNAS/lakka/system/atariosa.rom
ROM_OS_B_NTSC=/storage/roms/FreeNAS/lakka/system/atariosb.rom
ROM_OS_AA00R10=
ROM_OS_AA00R11=
ROM_OS_BB00R1=
ROM_OS_BB01R2=/storage/roms/FreeNAS/lakka/system/atarixl.rom
ROM_OS_BB02R3=
ROM_OS_BB02R3V4=
ROM_OS_CC01R4=
ROM_OS_BB01R3=
ROM_OS_BB01R4=
ROM_OS_BB01R59=
ROM_OS_BB01R59A=
ROM_5200=/storage/roms/FreeNAS/lakka/system/5200.rom
ROM_5200_A=
ROM_BASIC_A=
ROM_BASIC_B=
ROM_BASIC_C=
ROM_XEGAME=
ROM_400/800_CUSTOM=/storage/roms/FreeNAS/lakka/system/atariosb.rom
ROM_XL/XE_CUSTOM=
ROM_5200_CUSTOM=
ROM_BASIC_CUSTOM=
ROM_XEGAME_CUSTOM=
OS_400/800_VERSION=AUTO
OS_XL/XE_VERSION=AUTO
OS_5200_VERSION=AUTO
BASIC_VERSION=AUTO
XEGS_GAME_VERSION=AUTO
H1_DIR=
H2_DIR=
H3_DIR=
H4_DIR=
HD_READ_ONLY=1
PRINT_COMMAND=lpr %s
SCREEN_REFRESH_RATIO=1
MACHINE_TYPE=Atari XL/XE
RAM_SIZE=64
DEFAULT_TV_MODE=PAL
MOSAIC_RAM_NUM_BANKS=0
AXLON_RAM_NUM_BANKS=0
ENABLE_MAPRAM=0
DISABLE_BASIC=1
ENABLE_SIO_PATCH=1
ENABLE_SLOW_XEX_LOADING=0
ENABLE_H_PATCH=1
ENABLE_P_PATCH=1
ENABLE_NEW_POKEY=1
STEREO_POKEY=0
BUILTIN_BASIC=1
KEYBOARD_LEDS=0
F_KEYS=0
BUILTIN_GAME=0
KEYBOARD_DETACHED=0
1200XL_JUMPER=0
CFG_SAVE_ON_EXIT=0
MIO_ROM=
BLACK_BOX_ROM=
PROTO80_ROM=
CARTRIDGE_FILENAME=
CARTRIDGE_TYPE=0
CARTRIDGE_PIGGYBACK_FILENAME=
CARTRIDGE_PIGGYBACK_TYPE=0
CARTRIDGE_AUTOREBOOT=1
CASSETTE_FILENAME=
CASSETTE_LOADED=0
CASSETTE_WRITE_PROTECT=0
RTIME=1
XEP80_CHARSET=
AF80_ROM=
AF80_CHARSET=
COLOURS_NTSC_SATURATION=0
COLOURS_NTSC_CONTRAST=0
COLOURS_NTSC_BRIGHTNESS=0
COLOURS_NTSC_GAMMA=2.35
COLOURS_NTSC_HUE=0
COLOURS_NTSC_GTIA_DELAY=26.8
COLOURS_NTSC_EXTERNAL_PALETTE=
COLOURS_NTSC_EXTERNAL_PALETTE_LOADED=0
COLOURS_NTSC_ADJUST_EXTERNAL_PALETTE=0
COLOURS_PAL_SATURATION=0
COLOURS_PAL_CONTRAST=0
COLOURS_PAL_BRIGHTNESS=0
COLOURS_PAL_GAMMA=2.35
COLOURS_PAL_HUE=0
COLOURS_PAL_GTIA_DELAY=23.2
COLOURS_PAL_EXTERNAL_PALETTE=
COLOURS_PAL_EXTERNAL_PALETTE_LOADED=0
COLOURS_PAL_ADJUST_EXTERNAL_PALETTE=0
ARTIFACT_NTSC=NONE
ARTIFACT_PAL=NONE
ARTIFACT_NTSC_MODE=0
SCREEN_SHOW_SPEED=0
SCREEN_SHOW_IO_ACTIVITY=1
SCREEN_SHOW_IO_COUNTER=0
SCREEN_SHOW_1200XL_LEDS=1
FILTER_NTSC_SHARPNESS=-0.5
FILTER_NTSC_RESOLUTION=-0.1
FILTER_NTSC_ARTIFACTS=0
FILTER_NTSC_FRINGING=0
FILTER_NTSC_BLEED=0
FILTER_NTSC_BURST_PHASE=0
SOUND_ENABLED=1
SOUND_RATE=44100
SOUND_BITS=16
SOUND_BUFFER_MS=0
SOUND_LATENCY=20

ToKe79, 2018-04-04:
Loading ZIP from mounted network share currently does not work AFAIK.

RobLoach, 2018-04-04:
> Loading ZIP from mounted network share currently does not work AFAIK.

Works here... I used https://libreelec.wiki/how_to/mount_network_share .

jdca, 2018-04-05:
Got it wrong with my .atari800.cfg, removed it completly, that works too it seems. (read that it did not on the forum) Here's the new log.

This is up to i enter Memo Pad

[INFO] Using content: /storage/roms/FreeNAS/test/atari5200/Activision Decathlon, The (USA).a52.
[INFO] arg #0: retroarch
[INFO] arg #1: /storage/roms/FreeNAS/test/atari5200/Activision Decathlon, The (USA).a52
[INFO] arg #2: -s
[INFO] arg #3: /storage/savefiles
[INFO] arg #4: -S
[INFO] arg #5: /storage/savestates
[INFO] arg #6: -c
[INFO] arg #7: /storage/.config/retroarch/retroarch.cfg
[INFO] arg #8: -L
[INFO] arg #9: /tmp/cores/atari800_libretro.so
[INFO] Threaded video stats: Frames pushed: 405, Frames dropped: 0.
[INFO] Set config file to : /storage/.config/retroarch/retroarch.cfg
[INFO] RetroArch 1.7.1 (Git 29a057ac98)
[INFO] Redirecting save file to "/storage/savefiles/Activision Decathlon, The (USA).srm".
[INFO] Redirecting savestate to "/storage/savestates/Activision Decathlon, The (USA).state".
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Apr  3 2018
[INFO] Version: 1.7.1
[INFO] Git: 29a057ac98
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/atari800_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/Atari800/Atari800.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/Atari800/Activision Decathlon, The (USA).cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Atari800/Activision Decathlon, The (USA).cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Atari800/Activision Decathlon, The (USA).glslp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/Atari800/atari5200.cgp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/Atari800/atari5200.glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Atari800/Atari800.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Atari800/Atari800.glslp.
[INFO] Environ SET_CONTROLLER_INFO.
[INFO] Controller port: 1
[INFO]    ATARI Joystick (ID: 513)
[INFO]    ATARI Keyboard (ID: 259)
[INFO] Controller port: 2
[INFO]    ATARI Joystick (ID: 513)
[INFO]    ATARI Keyboard (ID: 259)
[INFO] Environ SET_VARIABLES.
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/Atari800/Activision Decathlon, The (USA).rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/Atari800/Atari800.rmp.
[INFO] Redirecting save file to "/storage/savefiles/Activision Decathlon, The (USA).srm".
[INFO] Redirecting savestate to "/storage/savestates/Activision Decathlon, The (USA).state".
[INFO] Environ GET_LOG_INTERFACE.
[INFO] Environ SYSTEM_DIRECTORY: "/storage/roms/FreeNAS/lakka/system".
[INFO] Environ CORE_ASSETS_DIRECTORY: "/storage/roms/downloads".
Retro SYSTEM_DIRECTORY /storage/roms/FreeNAS/lakka/system
Retro SAVE_DIRECTORY /storage/savefiles
Retro CONTENT_DIRECTORY /storage/roms/downloads
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Environ SET_INPUT_DESCRIPTORS:
[INFO] 	RetroPad, User 1, Button "B (bottom)" => "B"
[INFO] 	RetroPad, User 1, Button "Y (left)" => "Y"
[INFO] 	RetroPad, User 1, Button "Select" => "Select"
[INFO] 	RetroPad, User 1, Button "Start" => "Start"
[INFO] 	RetroPad, User 1, Button "D-Pad Up" => "Up"
[INFO] 	RetroPad, User 1, Button "D-Pad Down" => "Down"
[INFO] 	RetroPad, User 1, Button "D-Pad Left" => "Left"
[INFO] 	RetroPad, User 1, Button "D-Pad Right" => "Right"
[INFO] 	RetroPad, User 1, Button "A (right)" => "A"
[INFO] 	RetroPad, User 1, Button "X (up)" => "X"
[INFO] 	RetroPad, User 1, Button "L" => "L"
[INFO] 	RetroPad, User 1, Button "R" => "R"
[INFO] 	RetroPad, User 1, Button "L2" => "L2"
[... 273 lines cut ...]
[INFO] [udev]: Plugged pad: 8Bitdo NES30 Pro (11720:14368) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event2) supports 0 force feedback effects.
[INFO] [Autoconf]: 0 profiles found.
[INFO] [Autoconf]: no profiles found for 8Bitdo NES30 Pro (11720/14368).
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x2984f0
[INFO] Game focus is: : off.
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[libretro INFO] ChangeAV: w:336 h:240 ra:1.333333.

And this is after i loaded the rom manually.

Hack Libretro:crc A5200 ON sz:16 crc:f43e7cd0
Hack Libretro:A5200 cart->type:16 f43e7cd0

closed 2020-08-18 by Ntemis