libretro/Lakka-LibreELEC #427: MK808B-Plus graphic driver configuration issue on 2.1

trulio, 2018-03-23:
Hi,

when i boot my MK808B-Plus on image Lakka-S805.MK808B-Plus.arm-2.1.img.gz,

strorage resize behave normally but once retroarch launch screen buffer is corrupted

I tried on a samsung TV but also on a full HD BENQ computer monitor.

I have the same behavior.

I also tried the MXQ 2.1 image with the same results
(as kodi libreElec says MXQ and MK808 use the same image)

here are the files showing the issue:

Regards,

Trulio

[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/1842909/dmesg.txt)
[image]
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/1842910/log.txt)

gouchi, 2018-04-16:
Hi,

Do you reproduce the issue with [nightly builds](http://nightly.builds.lakka.tv/) ?

Thank you.

trulio, 2018-04-17:
Hello,

i just tried it and it works.

Do you know if bluetooth works? because it says service is enabled but vendor preset :  disabled

Thanks for your hard work.

Trulio

gouchi, 2018-04-17:
ping @Ntemis ;-)

Ntemis, 2018-04-18:
It has no bluetooth support ( fake advertising)

gouchi, 2018-04-18:
Thank you for the feedback.

So I will close the issue as it is running well with nightly.

closed 2018-04-18 by gouchi