libretro/Lakka-LibreELEC #429: Lakka-Generic.x86_64-2.1-devel-20180322000706-r28164-gc90e96b2e

jdca, 2018-03-24:
I wrote it to my usb-thumb, booted up, resized, rebooted again and now it just shows me the lakka logo "flower thing"

Running it on my Intel Nuc-i5

gouchi, 2018-03-24:
As written in the [template issue](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/ISSUE_TEMPLATE.md), 

> It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Thank you.

jdca, 2018-03-24:
Its important to be able to even access the computer before start logging also. Ssh is not on by default and when i press esc, screen turns black.

Maybe I could install Lakka-Generic.x86_64-2.1-devel-20180304204340-r28154-gb552a9a26, enable ssh and then upgrade to Lakka-Generic.x86_64-2.1-devel-20180322000706-r28164-gc90e96b2e and then log somehow? 

Thank you.

gouchi, 2018-03-24:
We should add it to the doc but you can boot with 
`live ssh`

To enable ssh.

jdca, 2018-03-24:
Just add it to the boot options with the live or install options.

Lakka:~ # systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.7.1 (Git 29a057ac98)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES 
Built: Mar 22 2018
[INFO] Version: 1.7.1
[INFO] Git: 29a057ac98
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 25 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 4: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 8: (1920x1080) 1920 x 1080, 25 Hz
[INFO] [DRM]: Mode 9: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 10: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 11: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 12: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 13: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 14: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 15: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 16: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 17: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 18: (720x576i) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 19: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 20: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 21: (720x480i) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 22: (720x480i) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 23: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 24: (640x480) 640 x 480, 60 Hz
[INFO] [GL]: Found GL context: �H������
�
[INFO] [GL]: Detecting screen resolution 0x0.
Segmentation fault (core dumped)

jdca, 2018-03-24:
Lakka:~ # dmesg
[    0.000000] microcode: microcode updated early to revision 0x20, date = 2017-01-27
[    0.000000] Linux version 4.11.12-rt14 (vudiq@helsinki) (gcc version 6.2.0 (GCC) ) #1 SMP PREEMPT RT Thu Mar 22 00:09:28 CET 2018
[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=2203-3454 disk=UUID=32f4ff39-3888-4b24-9bf3-0cbf0d97346a portable quiet vt.global_cursor_default=0 loglevel=2 ssh
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000c83effff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c83f0000-0x00000000c83f6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000c83f7000-0x00000000dba85fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dba86000-0x00000000dbb0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dbb10000-0x00000000dbb29fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dbb2a000-0x00000000dbc92fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dbc93000-0x00000000dbffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dbfff000-0x00000000dbffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd000000-0x00000000df1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000021fdfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI:                  /D54250WYK, BIOS WYLPT10H.86A.0021.2013.1017.1606 10/17/2013
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x21fe00 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-E7FFF uncachable
[    0.000000]   E8000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask 7E00000000 write-back
[    0.000000]   1 base 0200000000 mask 7FE0000000 write-back
[    0.000000]   2 base 00E0000000 mask 7FE0000000 uncachable
[    0.000000]   3 base 00DE000000 mask 7FFE000000 uncachable
[    0.000000]   4 base 00DD000000 mask 7FFF000000 uncachable
[    0.000000]   5 base 021FE00000 mask 7FFFE00000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: update [mem 0xdd000000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xdc000 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000fd6b0-0x000fd6bf] mapped at [ffff8800000fd6b0]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000097000] 97000 size 24576
[    0.000000] Using GB pages for direct mapping
[    0.000000] BRK [0x03cf4000, 0x03cf4fff] PGTABLE
[    0.000000] BRK [0x03cf5000, 0x03cf5fff] PGTABLE
[    0.000000] BRK [0x03cf6000, 0x03cf6fff] PGTABLE
[    0.000000] BRK [0x03cf7000, 0x03cf7fff] PGTABLE
[    0.000000] BRK [0x03cf8000, 0x03cf8fff] PGTABLE
[    0.000000] BRK [0x03cf9000, 0x03cf9fff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F0490 000024 (v02 INTEL )
[    0.000000] ACPI: XSDT 0x00000000DBB15080 000084 (v01 INTEL  D54250WY 00000015 AMI  00010013)
[    0.000000] ACPI: FACP 0x00000000DBB24A08 00010C (v05 INTEL  D54250WY 00000015 AMI  00010013)
[... 567 lines cut ...]
[   20.759606] Bluetooth: L2CAP socket layer initialized
[   20.759621] Bluetooth: SCO socket layer initialized
[   20.780174] usbcore: registered new interface driver btusb
[   20.820231] intel_rapl: Found RAPL domain package
[   20.820233] intel_rapl: Found RAPL domain core
[   20.820234] intel_rapl: Found RAPL domain uncore
[   20.820235] intel_rapl: Found RAPL domain dram
[   21.113605] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   21.149233] EXT4-fs (sda3): mounted filesystem with ordered data mode. Opts: (null)
[   21.185111] EXT4-fs (sda2): recovery complete
[   21.188219] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
[... cut ...]
[   34.898410] retroarch[699]: segfault at 3 ip 000000000050961f sp 00007f26902a3960 error 4 in retroarch[400000+2cc000]
[   51.149715] retroarch[709]: segfault at 3 ip 000000000050961f sp 00007f2dc7b6d960 error 4 in retroarch[400000+2cc000]
[   62.399006] retroarch[719]: segfault at 3 ip 000000000050961f sp 00007f0086960960 error 4 in retroarch[400000+2cc000]
[   66.147339] retroarch[726]: segfault at 3 ip 000000000050961f sp 00007f45d8ef0960 error 4 in retroarch[400000+2cc000]
[  103.283892] retroarch[732]: segfault at 3 ip 000000000050961f sp 00007fea8d57e960 error 4 in retroarch[400000+2cc000]
Lakka:~ #

jdca, 2018-03-24:
Lakka:~ # lspci 
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I218-V (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
Lakka:~ #

jdca, 2018-03-24:
Lakka:~ # lspci -nnk | grep -A3 VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation Haswell-ULT Integrated Graphics Controller [8086:0a26] (rev 09)
	Subsystem: Intel Corporation Haswell-ULT Integrated Graphics Controller [8086:2054]
	Kernel driver in use: i915
00:03.0 Audio device [0403]: Intel Corporation Haswell-ULT HD Audio Controller [8086:0a0c] (rev 09)

gouchi, 2018-03-24:
You graphic is supported but in your log you got ?
`[INFO] [GL]: Found GL context: �H����������`

It is workign with Lakka 2.1 ?

Thank you.

gouchi, 2018-03-24:
I reproduce the issue with an Intel HD 5500 with Lakka-Generic.x86_64-2.1-devel-20180322000706-r28164-gc90e96b2e.

`[INFO] [GL]: Found GL context: �H����������`

jdca, 2018-03-25:
Everything works with 

Lakka-Generic.x86_64-2.1-devel-20180304204340-r28154-gb552a9a26.img.gz

but not with 

Lakka-Generic.x86_64-2.1-devel-canary-d4d40fd.img

and 

Lakka-Generic.x86_64-2.1-devel-20180322000706-r28164-gc90e96b2e.img

gouchi, 2018-03-26:
Similar [issue](https://github.com/libretro/RetroArch/issues/6465) on RA issue tracker.

markwkidd, 2018-03-27:
This seems like the segmentation fault I'm getting with the March 22 build on my Asus eeePC 900A: https://gist.github.com/markwkidd/b97f63402e6a00a68094b7873e5b4267

autofire372, 2018-03-27:
Bug has been [fixed](https://github.com/libretro/RetroArch/commit/923ab3b6f4cf34e055f18b0d159270831db8935b) in RetroArch master.

closed 2018-03-31 by kivutar