libretro/Lakka-LibreELEC #430: consider migrating lakka docs to libretro/docs?

markwkidd, 2018-03-24:
From my perspective the new documentation system at https://github.com/libretro/docs has been a very big success.

There are several regular docs contributors, and the github/PR system means that there is a good way to track, approve, and when necessary undo changes to the docs.

Would the Lakka devs consider migrating the Lakka docs to this new system? With permission, I will volunteer to recreate the Lakka docs in the libretro/docs repository and others will probably help too.

RobLoach, 2018-03-25:
I think they look pretty nice where they are...
http://www.lakka.tv/doc/Home/

There are [a few places where the Wiki docs](https://github.com/libretro/Lakka-LibreELEC/wiki) could be consolidated, or moved/linked to libretro docs, but the Lakka-specific ones seem good where they are.

Hedonium, 2018-03-28:
The PPSSPP BIOS URL here is broken now. http://www.lakka.tv/doc/BIOSes/

Change to this? https://docs.libretro.com/library/ppsspp/#bios

kivutar, 2018-03-31:
@Hedonium done thanks.

kivutar, 2018-03-31:
@markwkidd the problem with merging the two docs, is that if we start deduplication pages, the users would see informations that doesn't apply to Lakka. They would see confusing information.

markwkidd, 2018-04-02:
Maybe there is a way to refactor both sets of docs better so that there is less redundancy -- that seems like a problem now because the two sets of docs overlap in their scope.

For some issues Lakka docs are more complete and sometimes the RetroArch docs are better. A few examples I can think of:

* The new RetroArch Keyboard Controls doc -- there's no process for backporting this new doc to Lakka docs or to crosslink the two docs so Lakka users can benefit.
* On the other hand, there are Lakka docs such as "Contributing your joypad config" and "Dynamic Wallpapers" that I think should also be in the RetroArch docs. Should I copy these over to the RA docs?  (probably I should...) Is it worth trying to keep the two docs versions synchronized once this content is also in the RA docs?

I don't know if there is an good solution but having more and more duplicated docs between Lakka and RetroArch seems somehow not good either.

I'm not sure!

kivutar, 2018-04-05:
> On the other hand, there are Lakka docs such as "Contributing your joypad config" and "Dynamic Wallpapers" that I think should also be in the RetroArch docs. Should I copy these over to the RA docs? [...]

I think it would be a good start yes.

There is no ideal solution to this problem.

closed 2020-08-18 by Ntemis