libretro/Lakka-LibreELEC #433: Scanned nes roms don't start.

jdca, 2018-03-28:
Lakka-RPi2.arm-2.1-devel-20180322014750-r28164-gc90e96b2e

I did a fresh install of lakka.
Scanned my roms wich i got from no-intro
Looked at my playlist

Nintendo - Nintendo Entertainment System.lpl
/storage/roms/FreeNAS/Favoriter/Nintendo/Nintendo Entertainment System/Battletoads-Double Dragon (USA).nes
Battletoads-Double Dragon (USA)
DETECT
DETECT
53B02F3D|crc

And when i tried to start it, nothing happened and my playlist look like this afterwards.

Nintendo - Nintendo Entertainment System.lpl
/storage/roms/FreeNAS/Favoriter/Nintendo/Nintendo Entertainment System/Battletoads-Double Dragon (USA).nes
Battletoads-Double Dragon (USA)
/tmp/cores/picodrive_libretro.so
Sega - MS/MD/CD/32X (PicoDrive)
53B02F3D|crc

Same happens with zipped roms.

markwkidd, 2018-04-02:
I think that this is fixed with the latest nightly versions.

If you are willing to try an experimental version, could you check to see if the problem still occurs with this build? http://nightly.builds.lakka.tv/2018-04-02/

jdca, 2018-04-02:
C64 doesnt work
FB Alpha doesnt work
GCE Vectrex doesnt work
NEC - PC - Engine TurboGrafX 16 works
NEC - PC Engine SuperGrafx doesnt work
Game Boy Advance works
Gameboy Color works
Gameboy works
Nintendo 64 doesnt work
Nintendo Entertainment System doesnt work
Super Nintendo (Hacks) works (But still gets that ugly black square instead of something nicer as an icon)
Super Nintendo doesnt work
Virtual boy doesnt work
Sega x32 doesnt work
Sega master system Mark III doesnt work
Sega mega driver soesnt work
SNK Neo Geo Color works
Sony playstation works

Every rom where I have to choose the core before it starts (the first time) works
The ones that starts emeditaly when pressing run doesnt work

C64 gets associated with dosbox_libretro.so
FB alpha with ColecoVision/SG-1000 (blueMSX)
Vectrex with bluemsx_libretro.so
PC Engine SuperGrafX with gambatte_libretro.so
Nintendo 64 with nestopia_libretro.so

And so on.

I was testing this on my raspberry pi 3 (not the newest pi 3) with Lakka-RPi2.arm-2.1-devel-20180402004820-r28168-g3c449c9d4.img.gz

jdca, 2018-04-04:
Lakka-RPi2.arm-2.1-devel-20180404001655-r28169-gbbabe7b3c
Rpi3

It works as it should now, you can close this thread if you ask me.

ToKe79, 2018-04-04:
closed per user request

closed 2018-04-04 by ToKe79