libretro/Lakka-LibreELEC #438: Legacy BIOS (2005) - USB mass storage device not seen as bootable

Golffies, 2018-04-02:
According to Gparted, Lakka image comes with a GPT + protective MBR partition table. It does not boot on my 10+ years old PC, equipped with 2005' BIOS from Award. The BIOS does have an option to boot from USB flash memory.

Converting the partition table [from GPT to MBR](http://www.rodsbooks.com/gdisk/mbr2gpt.html) on the flashed USB storage device does not help.

Trying to flash the USB storage device with tools such as [UNetbootin](http://unetbootin.github.io/) instead of regular DD tool does not help neither. 

- Lakka : V 2.1
- Hardware : Award BIOS (2005 - most recent release) booting an Asus mPGA478 motherboard
- Step by step :

1. Extracting Lakka 2.1 image;
2. Flashing image to a 1GB USB flash storage memory using DD cli application;
3. Mounting the `LAKKA` and `LAKKA STORAGE` partitions;
4. Checking both partitions file system using Gparted;
5. Rebooting;
6. Entering the BIOS setup;
7. Choosing `USB Zip/Flash` as the first device to boot from;
8. Exiting the BIOS;
9. Rebooting;
10. Getting a `Boot error` prompt from the BIOS on the same screen as the POST messages; i.e. the prompt does come from the BIOS, not from Syslinux.

- Expected result : Syslinux should have been launched
- Observed result : the boot code of the MBR fails to launch Syslinux

markwkidd, 2018-04-02:
Do any other operating systems successfully boot from that USB flash drive? Also, have you tried another USB flash drive with Lakka?

I have a bunch of old 1GB flash drives that don't boot Lakka correctly for some reason. That is the reason I ask.

retorquere, 2018-04-14:
I may be seeing the same thing. I can install openelec from that USB drive, but with Lakka on the same USB, it just boots past it. Tried setting removable device as first in bios, and explicitly selecting the Lakka USB drive from the boot menu during boot, but it just boots through to the existing OpenELEC install.

gouchi, 2018-04-14:
You may try to use [Plop boot manager](https://www.plop.at/en/bootmanager/plpbt.bin.html) to boot on usb.

retorquere, 2018-04-14:
Can this be on the same usb, or on a second stick?

gouchi, 2018-04-14:
second stick (or cd, windows boot menu,  network pxe boot...)

retorquere, 2018-04-15:
plop does work, but when I boot through to USB, I get a black screen with a blinking cursor. Tried with 3 different USB sticks, each different brands.

gouchi, 2018-04-15:
Can you try to flash your usb stick with [Etcher](https://etcher.io/) ?

Thank you.

Golffies, 2018-08-03:
@gouchi : Plop boot manager actually let me boot on USB. On a daily basis it is not a convenient fix. Flashing the USB stick with Etcher did not improve things.

@markwkidd : around 10 years ago, I successfully boot the very same computer from the very same USB stick (resilient pieces of hardware, aren't they ?) [with Geexbox 1.x](http://www.junauza.com/2007/10/geexbox-windows-media-center-killer.html) for i686 platform.

_[Geexbox legacy iso](http://download.geexbox.org/releases/)._

However, I do not remember me booting the PC from USB with another distro. I tried to boot Lakka from a 64GB USB stick on the same PC, with no more success. Obviously, the BIOS on this machine has limited features; it likely expects a basic MBR to be present on the USB stick, which Lakka image fails to provide. Should a next version embed a more compatible MBR, it would be great.

gouchi, 2018-08-03:
You can [install Plop to your mbr ](https://www.plop.at/en/bootmanager/mbrinstall.html)then boot on your disk or usb stick. It is a nice boot manager to have ;-)

closed 2020-08-18 by Ntemis