libretro/Lakka-LibreELEC #439: Bug in PPSSPP in RetroArch (Lakka Nightly) (Generic X86_64) When playing GTA Vice City Stories

RetroFan90, 2018-04-02:
I Am running into an issue with the PPSSPP core in the latest Lakka Nightly Build.
The issue is noticable when i compare it to the nightly build of PPSSPP (Version 1.4.2-git <<Lakka Version)(Currently 1.5.4 against the Windows X64 version.)
I'm left with the conclusion that the nightly version isn't keeping up with the newer cores.
[image]
[image]
The Picture with the garbled lines is from Lakka, whereas the Clean Image is from the Windows Version 1.5.4
Please Look Into This. :)

ToKe79, 2018-04-04:
Hi,

I [updated the core](https://github.com/ToKe79/Lakka-LibreELEC/commit/99a12be9215517bdec0bdddfff7f8f3947d517d5) to the latest and started build of all targets. You can download the images from here:
http://nightly.builds.lakka.tv/special_builds/ppsspp-update/

Currently there is only image for Generic.x86_64, but the remaining images  should appear as soon as they built.

For the future, please fill out as much as possible information in the [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/ISSUE_TEMPLATE.md) that is shown when opening new issue.

Please report your test results back in this thread. Thank you!

Ntemis, 2018-04-30:
This is not lakka related closing down,
@RetroFan90 please report the issue here https://github.com/libretro/ppsspp
as this is a bug clearly about standalone vs libretro core

closed 2018-04-30 by Ntemis