libretro/Lakka-LibreELEC #441: MSX and MSX2 roms on Lakka-RPi2.arm-2.1-devel won't start

jdca, 2018-04-03:
I did a thread about this 2 weeks ago wich was closed as fixed, I couldn't reopen that thread for some reason?

Anyways, the issue still exist with "content" instead of "rom"

Lakka-RPi2.arm-2.1-devel-20180402004820-r28168-g3c449c9d4
MSX and MSX2 roms scans but doesn't start. "failed to load content"

msx

[INFO] Updating firmware status for: /tmp/cores/bluemsx_libretro.so on /storage/roms/FreeNAS/lakka/system
[INFO] Using content: /storage/roms/FreeNAS/scans-ok/Microsoft/MSX/Confused (Europe).zip.
[INFO] arg #0: retroarch
[INFO] arg #1: /storage/roms/FreeNAS/scans-ok/Microsoft/MSX/Confused (Europe).zip
[INFO] arg #2: -s
[INFO] arg #3: /storage/savefiles
[INFO] arg #4: -S
[INFO] arg #5: /storage/savestates
[INFO] arg #6: -c
[INFO] arg #7: /storage/.config/retroarch/retroarch.cfg
[INFO] arg #8: -L
[INFO] arg #9: /tmp/cores/bluemsx_libretro.so
[INFO] Threaded video stats: Frames pushed: 1638, Frames dropped: 0.
[INFO] Set config file to : /storage/.config/retroarch/retroarch.cfg
[INFO] RetroArch 1.7.1 (Git 29a057ac98)
[INFO] Redirecting save file to "/storage/savefiles/Confused (Europe).srm".
[INFO] Redirecting savestate to "/storage/savestates/Confused (Europe).state".
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4
Built: Apr 1 2018
[INFO] Version: 1.7.1
[INFO] Git: 29a057ac98
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/bluemsx_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/blueMSX/blueMSX.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/blueMSX/Confused (Europe).cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/blueMSX/Confused (Europe).cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/blueMSX/Confused (Europe).glslp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/blueMSX/MSX.cgp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/blueMSX/MSX.glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/blueMSX/blueMSX.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/blueMSX/blueMSX.glslp.
[INFO] Environ SET_VARIABLES.
[INFO] Environ SET_CONTROLLER_INFO.
[INFO] Controller port: 1
[INFO] RetroPad (ID: 1)
[INFO] RetroKeyboard (ID: 3)
[INFO] RetroPad Keyboard Map (ID: 513)
[INFO] Controller port: 2
[INFO] RetroPad (ID: 1)
[INFO] RetroKeyboard (ID: 3)
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/blueMSX/Confused (Europe).rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/blueMSX/blueMSX.rmp.
[INFO] Redirecting save file to "/storage/savefiles/Confused (Europe).srm".
[INFO] Redirecting savestate to "/storage/savestates/Confused (Europe).state".
[INFO] Environ GET_LOG_INTERFACE.
[INFO] Content loading skipped. Implementation will load it on its own.
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Environ GET_VARIABLE bluemsx_msxtype:
[INFO] Auto
[INFO] Environ GET_VARIABLE bluemsx_overscan:
[INFO] disabled
[INFO] Environ GET_VARIABLE bluemsx_vdp_synctype:
[INFO] Auto
[INFO] Environ GET_VARIABLE bluemsx_nospritelimits:
[INFO] OFF
[INFO] Environ GET_VARIABLE bluemsx_ym2413_enable:
[INFO] enabled
[INFO] Environ GET_VARIABLE bluemsx_cartmapper:
[... 237 lines cut ...]
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event0).
[INFO] [udev]: Plugged pad: 8Bitdo NES30 Pro (11720:14368) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event2) supports 0 force feedback effects.
[INFO] [Autoconf]: 0 profiles found.
[INFO] [Autoconf]: no profiles found for 8Bitdo NES30 Pro (11720/14368).
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x2984f0
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].

ToKe79, 2018-04-04:
Do you have write permission in the folder? I guess you load the content from mounted NAS (via samba?) - AFAIK there is an issue with loading ZIP files from mounted samba share, maybe this is related.

jdca, 2018-04-04:
Yeah it's full read write excute, i moved them to my rpi3 and unzipped them, still doesnt load.

msx

[INFO] Using content: /storage/roms/MSX/Antarctic Adventure (Europe).rom.
[INFO] arg #0: retroarch
[INFO] arg #1: /storage/roms/MSX/Antarctic Adventure (Europe).rom
[INFO] arg #2: -s
[INFO] arg #3: /storage/savefiles
[INFO] arg #4: -S
[INFO] arg #5: /storage/savestates
[INFO] arg #6: -c
[INFO] arg #7: /storage/.config/retroarch/retroarch.cfg
[INFO] arg #8: -L
[INFO] arg #9: /tmp/cores/bluemsx_libretro.so
[INFO] Threaded video stats: Frames pushed: 342, Frames dropped: 0.
[INFO] Set config file to : /storage/.config/retroarch/retroarch.cfg
[INFO] RetroArch 1.7.1 (Git 29a057ac98)
[INFO] Redirecting save file to "/storage/savefiles/Antarctic Adventure (Europe).srm".
[INFO] Redirecting savestate to "/storage/savestates/Antarctic Adventure (Europe).state".
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Apr  3 2018
[INFO] Version: 1.7.1
[INFO] Git: 29a057ac98
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/bluemsx_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/blueMSX/blueMSX.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/blueMSX/Antarctic Adventure (Europe).cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/blueMSX/Antarctic Adventure (Europe).cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/blueMSX/Antarctic Adventure (Europe).glslp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/blueMSX/MSX.cgp.
[INFO] Shaders: no parent-dir-specific preset found at /tmp/shaders/presets/blueMSX/MSX.glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/blueMSX/blueMSX.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/blueMSX/blueMSX.glslp.
[INFO] Environ SET_VARIABLES.
[INFO] Environ SET_CONTROLLER_INFO.
[INFO] Controller port: 1
[INFO]    RetroPad (ID: 1)
[INFO]    RetroKeyboard (ID: 3)
[INFO]    RetroPad Keyboard Map (ID: 513)
[INFO] Controller port: 2
[INFO]    RetroPad (ID: 1)
[INFO]    RetroKeyboard (ID: 3)
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/blueMSX/Antarctic Adventure (Europe).rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/blueMSX/blueMSX.rmp.
[INFO] Redirecting save file to "/storage/savefiles/Antarctic Adventure (Europe).srm".
[INFO] Redirecting savestate to "/storage/savestates/Antarctic Adventure (Europe).state".
[INFO] Environ GET_LOG_INTERFACE.
[INFO] Content loading skipped. Implementation will load it on its own.
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Environ GET_VARIABLE bluemsx_msxtype:
[INFO] 	Auto
[INFO] Environ GET_VARIABLE bluemsx_overscan:
[INFO] 	disabled
[INFO] Environ GET_VARIABLE bluemsx_vdp_synctype:
[INFO] 	Auto
[INFO] Environ GET_VARIABLE bluemsx_nospritelimits:
[INFO] 	OFF
[INFO] Environ GET_VARIABLE bluemsx_ym2413_enable:
[INFO] 	enabled
[INFO] Environ GET_VARIABLE bluemsx_cartmapper:
[INFO] 	Auto
[INFO] Environ SYSTEM_DIRECTORY: "/storage/roms/FreeNAS/lakka/system".
[ERROR] Failed to load content.
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/Antarctic Adventure (Europe).srm".
[INFO] Redirecting savestate to "/storage/savestates/Antarctic Adventure (Europe).state".
[... 230 lines cut ...]
[INFO] [udev]: Plugged pad: 8Bitdo NES30 Pro (11720:14368) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event2) supports 0 force feedback effects.
[INFO] [Autoconf]: 0 profiles found.
[INFO] [Autoconf]: no profiles found for 8Bitdo NES30 Pro (11720/14368).
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x2984f0
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].

roms are from no-intro

closed 2020-08-18 by Ntemis