libretro/Lakka-LibreELEC #443: MAME.zip

jdca, 2018-04-04:
http://thumbnailpacks.libretro.com/MAME.zip seems to be broken
27-Mar-2018 01:22          7121167416
md5sum MAME.zip 808629ba5f761f72763aaaeb3cdba2b9

jdca, 2018-04-04:
Personally i don't care much of the mame, just noticed when trying to unzip them along with all the other thumbnails so i figured i would tell about the issue.

What the problem is i don't know, couldn't unzip them for some reason.

RobLoach, 2018-04-05:
What OS are you on? Mind trying again? Looks like the packs were recreated.

jdca, 2018-04-05:
Antergos/Arch and i did a wget on all files, unzipped all at once when they were ready, just saw that mame didnt not work. Downloaded it again and tried one more time, didnt unzip.

This time i tried unzip in cli and that worked, unzip in nautilus/gnome did not. 
You can close this, i'll have to tell someone else about this.

Sorry for this, I should have done some more testing.

RobLoach, 2018-04-05:
Glad it's working now! Considering the size of the zip, I'm not surprised it didn't work in Nautilus.

closed 2018-04-05 by RobLoach