libretro/Lakka-LibreELEC #447: PC x86: No driver for Radeon HD 3450 (Couldn't find a suitable DRM device.)

RollingStar, 2018-04-08:
- Which version of Lakka are you using? 2.1 stable (no apparent driver change in CHANGELOG)
- What did you do? Installed Lakka and booted the PC
- What did you expect to happen? Video to be displayed
- What happened instead? No video

System works fine when outputting VGA from the main motherboard.

System:

- [Dell Optiplex 780 SFF](https://www.dell.com/downloads/global/products/optix/en/optiplex_780_tech_guidebook_en.pdf) (yes, 780, I don't know why the log says 980)
- Intel® Core™ 2 Duo E7500/2.93GHz, 3M, 1066FSB (or similar)
- [Radeon HD 3450 low profile (with S-video and DMS-59 ports)](https://www.amazon.com/ATI-Radeon-Express-DMS-59-Profile/dp/B004QPR04O)

```
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (official): 2.1 (Generic.i386)
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.6.9 (Git c056726)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4
Built: Nov 21 2017
[INFO] Version: 1.6.9
[INFO] Git: c056726
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/r
troarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state
.
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template
glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)
Lakka:~ # lspci -nnk | grep -A 3 VGA
01:00.0 VGA compatible controller [0300]: Advanced Micro De
ices, Inc. [AMD/ATI] RV620 LE [Radeon HD 3450] [1002:95c5]
        Subsystem: Dell RV620 LE [Radeon HD 3450] (OptiPlex
980) [1028:0342]
Lakka:~ # for p in /sys/class/drm/*/status; do con=${p%/sta
us}; echo -n "${con#*/card?-}: "; cat $p; done
/sys/class/drm/*: cat: can't open '/sys/class/drm/*/status'
 No such file or directory
Lakka:~ # ls -l /sys/class/drm/*
lrwxrwxrwx    1 root     root             0 Apr  1 00:38 /s
s/class/drm/ttm -> ../../devices/virtual/drm/ttm
-r--r--r--    1 root     root          4096 Apr  1 00:38 /s
s/class/drm/version
Lakka:~ # ls -l /sys/class/drm/*/
total 0
drwxr-xr-x    2 root     root             0 Apr  1 00:51 po
er
lrwxrwxrwx    1 root     root             0 Apr  1 00:38 su
system -> ../../../../class/drm
[... 237 lines cut ...]
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/r
troarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state
.
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template
glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)
Lakka:~ #
```

gouchi, 2018-04-08:
Please provide the [graphic card log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#graphic-card-logs).

Thank you.

gouchi, 2018-04-08:
Hi,

Both cards are [supported](https://cateee.net/lkddb/web-lkddb/DRM_RADEON.html) (1002:95c5 and 1002:6759) by radeon kernel driver.

Can you try to make a test with [nightly build](http://nightly.builds.lakka.tv) if you reproduce the issue ?

Also you can try 64 bit image.

RollingStar, 2018-04-08:
Could I test from an x64 live USB or would I have to install it to be sure?

ToKe79, 2018-04-09:
Hi,
you can try live boot from USB.

ToKe79, 2018-06-21:
@RollingStar any luck with the nightly build?

ToKe79, 2019-06-26:
same as #543, fix comming soon

closed 2019-06-26 by ToKe79