libretro/Lakka-LibreELEC #449: In compiling the Lakka code, this problem always appears on the PPSSPP file, hoping to find help.

luojinhui, 2018-04-11:
UNPACK   ppsspp
Cloning into '/home/luojh/Lakka-LibreELEC/build.Lakka-ROCK64.arm-2.1-devel/ppsspp-dd73f91'...
remote: Counting objects: 145048, done.
remote: Compressing objects: 100% (79/79), done.
remote: Total 145048 (delta 44), reused 47 (delta 29), pack-reused 144940
Receiving objects: 100% (145048/145048), 101.06 MiB | 78.00 KiB/s, done.
Resolving deltas: 100% (113106/113106), done.
Checking connectivity... done.
error: pathspec 'dd73f91' did not match any file(s) known to git.
make: *** [image] Error 1

luojinhui, 2018-04-11:
my project is Rockchip device is ROCK64

luojinhui, 2018-04-11:
It means that we modify PKG_VERSION for 7b61523 in /ppsspp/package.mk？？
I tried, but I still have problems.
like this:
Assembler messages:
Error: unrecognized option -mabi=aapcs-linux
make[1]: *** [Makefile:432: ../ext/native/math/fast/fast_matrix_neon.o] Error 1
make[1]: *** Waiting for unfinished jobs....
../ext/zlib/gzread.c: In function 'gz_load':
../ext/zlib/gzread.c:30:15: warning: implicit declaration of function 'read' [-Wimplicit-function-declaration]
         ret = read(state->fd, buf + *have, len - *have);
               ^~~~
../ext/zlib/gzread.c: In function 'gzclose_r':
../ext/zlib/gzread.c:595:11: warning: implicit declaration of function 'close' [-Wimplicit-function-declaration]
     ret = close(state->fd);
           ^~~~~
make[1]: Leaving directory '/home/luojh/Lakka-LibreELEC/build.Lakka-ROCK64.arm-2.1-devel/ppsspp-7b61523/libretro'
make: *** [image] Error 2

What can I do to compile it?

RobLoach, 2018-04-11:
It's possible that https://github.com/libretro/ppsspp is broken.

SapphireTactician, 2018-04-20:
@luojinhui As a workaround for the time being, assuming you don't mind using Lakka without the PPSSPP core, you can just remove `ppsspp` from the `LIBRETRO_CORES=` string in `/distribution/Lakka/options`

gouchi, 2018-04-29:
Hi,

It  should be fixed using latest commit.

Can you make a test with [nightly builds](http://nightly.builds.lakka.tv/2018-04-29/Rockchip.ROCK64.arm/Lakka-Rockchip.ROCK64.arm-2.1-devel-20180429005234-r28180-g790ccef8f.img.gz) ?

Thank you.

luojinhui, 2018-04-29:
@gouchi Sure,I will be complete this test and reply to you as soon as possible

ToKe79, 2018-05-14:
Fixed with #445

closed 2018-05-14 by ToKe79