libretro/Lakka-LibreELEC #452: Abnormal Wiimote Plus (RVL-036) Behavior

SapphireTactician, 2018-04-19:
### Description:

When pairing / connecting a Wii Remote Plus (*RVL-036*), left and right directions are also treated as up and down. When trying to bind the inputs manually to correct this, the changes will register but will be ignored / not be respected. Attempting to bind any buttons aside from the d-pad such as `1`, `2`, `A`, `B`, etc. fail as the bind simply times out.

I've test this with 3 different Wiimotes, all of which are official Nintendo brand.

In a rather interesting discovery, the workaround to this issue is to select `Quit RetroArch` (*which automatically restarts RA as well. Using Reboot WON'T work*) after the Wii Remote Plus has paired / connected. This is not a permanent workaround and must be performed every time you pair or connect a Wiimote / reboot Lakka. 

*****

### Specs:

**Device** Raspberry Pi 3B+    

**Lakka Version:** devel-20180415193509-r28172-gaf3f01ae4 (_RPi2.arm_) (_Locally built canary_)  

**RetroArch Version:** 1.7.2 (*Git 6d00810716*)

-------

### Expected Behavior:

The wiimote should behave properly after connecting and being acknowledged by the auto-config.  

-------

### Actual Behavior:

Left and Right directions are also treated as Up and Down. Rebinding the directions registers but is otherwise ignored. All other inputs aren't recognized at all during rebinding only and will time out during binding.  

The issue is temporarily resolved if the user quits (*and restarts*) RetroArch after a Wiimote has already been connected via pairing and connecting via `bluetoothctl` or pressing both 1+2 after one has already been paired. Rebooting Lakka does not have this effect. The workaround stops works after Lakka has been shutdown or rebooted until the workaround is performed again.

-------

### Steps to Reproduce:

* Pair / connect a Wii Remote Plus (_may also affect the original Wiimote but I do not have one to confirm or deny that_)
* Press left or right and observe that it moves up and down instead
* Attempt to rebind inputs to observe other issues
* Use `Quit RetroArch` to observe workaround
    * Work around remains in effect until either Lakka is rebooted or the controller is reconnected / re-paired

-------

### Logs:

This is all I could acquire from my terminal. I can't identify anything out of the ordinary here but I'll share them for completion's sake.

**Verbose Output - Before applying workaround:**  
[Lakka-VerboseLogs_Before-Workaround_20180418-1930.txt](https://github.com/libretro/Lakka-LibreELEC/files/1926256/Lakka-VerboseLogs_Before-Workaround_20180418-1930.txt)  

**Verbose Output - Pairing the RVL-036:**  
[Lakka-VerboseLogs_Pairing.txt](https://github.com/libretro/Lakka-LibreELEC/files/1926258/Lakka-VerboseLogs_Pairing.txt)  

**Verbose Output - After applying workaround:**  
[Lakka-VerboseLogs_After-Workaround_20180418-1937.txt](https://github.com/libretro/Lakka-LibreELEC/files/1926257/Lakka-VerboseLogs_After-Workaround_20180418-1937.txt)  

**Verbose Output - `dmesg -r`**  
[Lakka-VerboseLogs_dmesg-r.txt](https://github.com/libretro/Lakka-LibreELEC/files/1926259/Lakka-VerboseLogs_dmesg-r.txt)

gouchi, 2018-04-19:
First, can you try to launch in the [Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/) 
`udevadm monitor`

To check when the [udev rules](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/packages/libretro/retroarch/udev.d/99-nintendo-wii-remote.rules) is apply for the wii remote before or after you restart RetroArch.

Thank you.

SapphireTactician, 2018-04-21:
@gouchi  It looks like before IF I'm reading this output correctly:

[udevadm_terminal_output.txt](https://github.com/libretro/Lakka-LibreELEC/files/1934029/udevadm_terminal_output.txt)

gouchi, 2018-04-25:
Maybe we will have to add an [udev rule](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/packages/libretro/retroarch/udev.d/99-nintendo-wii-remote.rules)

Can you try to add udev rule to` /storage/.config/udev.rules.d ` ?

`nano /storage/.config/udev.rules.d/99-wii-remote-plus.rules` then add this content

`SUBSYSTEM=="input", ATTRS{idVendor}=="1406", ATTRS{idProduct}=="774", MODE="666", ENV{ID_INPUT_JOYSTICK}="1"`

Then reload udev rules
`udevadm control --reload-rules && udevadm trigger`

Thank you

SapphireTactician, 2018-04-27:
@gouchi Sorry for the slow response. I generated the udev rule as advised, and reloaded the rules. Sadly, there's no improvement. I tried rebooting the device as well, still no improvement.

What would you like me to try next?

gouchi, 2018-04-29:
Thank you for your tests.

Is it possible to get udevadm monitor and RA log when you added the udev rule ?

Thank you.

SapphireTactician, 2018-05-03:
Idk if it's the new udev rules or not but I can't even connect anymore. I just get the following

```
Lakka:~ # udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[34.923770] add      /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
UDEV  [34.929090] add      /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
KERNEL[35.230327] remove   /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
UDEV  [35.233991] remove   /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
```

gouchi, 2018-05-16:
If there is an joypad autoconfiguration file in (/tmp/joypads/udev) related to Wii Remote, please try to rename it.

[Source](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6)

SapphireTactician, 2018-05-20:
I'll attempt this when I have more time. Sorry for making you wait.

closed 2020-08-18 by Ntemis