libretro/Lakka-LibreELEC #463: OSD Font

RobLoach, 2018-04-29:
Should have DejaVu mono install correctly.

From RetroArch's `freetype.c`.....
```
static const char *font_paths[] = {
#if defined(_WIN32)
   "C:\\Windows\\Fonts\\consola.ttf",
   "C:\\Windows\\Fonts\\verdana.ttf",
#elif defined(__APPLE__)
   "/Library/Fonts/Microsoft/Candara.ttf",
   "/Library/Fonts/Verdana.ttf",
   "/Library/Fonts/Tahoma.ttf",
#else
   "/usr/share/fonts/TTF/DejaVuSansMono.ttf",
   "/usr/share/fonts/TTF/DejaVuSans.ttf",
   "/usr/share/fonts/truetype/ttf-dejavu/DejaVuSansMono.ttf",
   "/usr/share/fonts/truetype/ttf-dejavu/DejaVuSans.ttf",
   "/usr/share/fonts/truetype/dejavu/DejaVuSansMono.ttf",
   "/usr/share/fonts/truetype/dejavu/DejaVuSans.ttf",
   "/usr/share/fonts/TTF/Vera.ttf",
   "/usr/share/fonts/truetype/droid/DroidSansFallbackFull.ttf",
#endif
   "osd-font.ttf", /* Magic font to search for, useful for distribution. */
};
```

RobLoach, 2018-06-26:
This should be fixed with an update to both `retroarch` and `retroarch-assets`.

Ntemis, 2018-06-26:
When new ra version is out.

kivutar, 2018-08-12:
Is the current OSD font broken?

We're forcing the same font used by xmb in retroarch.cfg

RobLoach, 2018-08-12:
It may be fixed, but haven't tested lately. Before it was appearing as monospace.

kivutar, 2018-08-24:
I confirm it is fixed

closed 2018-08-24 by kivutar