libretro/Lakka-LibreELEC #473: Wifi not working in 64bit image

falco1029, 2018-05-19:
I'm running Lakka 2.1 on a Dell Latitude D630, and when using the 64-bit Image only, it won't allow me to use Wifi. I get only a jumble of characters in one entry on the wifi list, and it won't connect. 

It does, however, let me use Wired without any issue. Using wired to try enabling wifi via SSH fails, though, giving me "Method "Scan" with signature "" on interface "net.connman.technology"  doesn't exist"

The 32-bit image works fine, but I wanted to switch over to the 64 bit so I could use the better Saturn emulator and N64 emulator.

Log:
```
[INFO] RetroArch 1.6.9 (Git c056726)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3
Built: Nov 21 2017
[INFO] Version: 1.6.9
[INFO] Git: c056726
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 4 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 1 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Connector 3 connected: no
[INFO] [DRM]: Connector 3 has 0 modes.
[INFO] [DRM]: Mode 0: (1280x800) 1280 x 800, 60 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1280x800.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x7f584c1b8c80.
[INFO] [KMS]: New FB: 1280x800 (stride: 5120).
[INFO] [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 965GM .
[INFO] [GL]: Version: 2.1 Mesa 17.1.3.
[INFO] [GL]: Using resolution 1280x800
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Using GL_RGB565 for texture uploads.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event4).
[INFO] [udev]: Mouse #0 (/dev/input/event5).
[INFO] [udev]: Mouse #1 (/dev/input/mouse0).
[INFO] [udev]: Mouse #0 (/dev/input/event6).
[ERROR] [udev] Failed to open device: /dev/input/mouse1 (Inappropriate ioctl for device).
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 1024 frames
[INFO] ALSA: Buffer size: 2048 frames
[INFO] [Menu]: Found menu display driver: "menu_display_gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [KMS]: New FB: 1280x800 (stride: 5120).
[INFO] [KMS]: New FB: 1280x800 (stride: 5120).
```

gouchi, 2018-05-21:
Do you reproduce the issue with our [nightly builds](http://nightly.builds.lakka.tv/) ?

Thank you.

falco1029, 2018-05-21:
Yes, issue persists in nightly build.

ToKe79, 2019-07-02:
no response

closed 2019-07-02 by ToKe79