libretro/Lakka-LibreELEC #478: Mounting of local storage fails after rebooting

TeamKinetic, 2018-05-29:
Hi all,

I've got an intel based mini PC that has OpenElec/kodi installed. Using Lakka 2.1 from USB.

I am trying to use Lakka from a USB drive and running into some issues. It all boots up the first time no bother and mounts the local hard drive partition under the storage folder.

If I shutdown and remove the USB drive and boot from the harddrive into OpenElec at any time the Lakka USB drive will not mount the local storage anymore. Please see the output from dmesg below.

If I recreate the USB image and boot again all is fine.

Lakka:~ # dmesg
[    0.000000] microcode: microcode updated early to revision 0xa0b, date = 2010-09-28
[    0.000000] Linux version 4.11.12-rt14 (kivutar@Server01) (gcc version 6.2.0 (GCC) ) #1 SMP PREEMPT RT Sat Nov 18 21:12:49 CST 2017
[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=2111-3926 disk=UUID=b9ef8ad9-785d-4546-a992-032454f63ce7 portable quiet vt.global_cursor_default=0 loglevel=2
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000006ff9ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000006ffa0000-0x000000006ffadfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000006ffae000-0x000000006ffeffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000006fff0000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff700000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: AOpen   DE7000/nMCP7ALPx-DE R1.04 Oct.03.2011, BIOS 080015  10/03/2011
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x6ffa0 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-DFFFF uncachable
[    0.000000]   E0000-EFFFF write-through
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask F80000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] found SMP MP-table at [mem 0x000ff780-0x000ff78f] mapped at [ffff8800000ff780]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000099000] 99000 size 24576
[    0.000000] BRK [0x03cf3000, 0x03cf3fff] PGTABLE
[    0.000000] BRK [0x03cf4000, 0x03cf4fff] PGTABLE
[    0.000000] BRK [0x03cf5000, 0x03cf5fff] PGTABLE
[    0.000000] BRK [0x03cf6000, 0x03cf6fff] PGTABLE
[    0.000000] BRK [0x03cf7000, 0x03cf7fff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000FA970 000014 (v00 ACPIAM)
[    0.000000] ACPI: RSDT 0x000000006FFA0000 000044 (v01 100311 RSDT1105 20111003 MSFT 00000097)
[    0.000000] ACPI: FACP 0x000000006FFA0200 000084 (v01 100311 FACP1105 20111003 MSFT 00000097)
[    0.000000] ACPI: DSDT 0x000000006FFA04A0 0094A8 (v01 NGBAO  NGBAO104 00000104 INTL 20051117)
[    0.000000] ACPI: FACS 0x000000006FFAE000 000040
[    0.000000] ACPI: APIC 0x000000006FFA0390 000080 (v01 100311 APIC1105 20111003 MSFT 00000097)
[    0.000000] ACPI: MCFG 0x000000006FFA0410 00003C (v01 100311 OEMMCFG  20111003 MSFT 00000097)
[    0.000000] ACPI: WDRT 0x000000006FFA0450 000047 (v01 100311 NV-WDRT  20111003 MSFT 00000097)
[    0.000000] ACPI: OEMB 0x000000006FFAE040 00007A (v01 100311 OEMB1105 20111003 MSFT 00000097)
[... 622 lines cut ...]
[   26.586425] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[   26.587068] systemd[1]: Created slice System Slice.
[   26.762495] systemd-journald[264]: Received request to flush runtime journal from PID 1
[   27.589072] ACPI: PCI Interrupt Link [LAZA] enabled at IRQ 23
[   27.589086] snd_hda_intel 0000:00:08.0: Disabling MSI
[   27.828347] intel_powerclamp: No package C-state available
[   27.858316] r8169 0000:04:00.0 eth0: link down
[   27.858419] r8169 0000:04:00.0 eth0: link down
[   27.858708] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   28.033567] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC662 rev1: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:hp
[   28.033571] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   28.033573] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   28.033575] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   28.033576] snd_hda_codec_realtek hdaudioC0D0:    dig-out=0x1e/0x0
[   28.033578] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   28.033580] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x18
[   28.033582] snd_hda_codec_realtek hdaudioC0D0:      Line=0x1a
[   28.727875] input: HDA NVidia Mic as /devices/pci0000:00/0000:00:08.0/sound/card0/input10
[   28.727996] input: HDA NVidia Line as /devices/pci0000:00/0000:00:08.0/sound/card0/input11
[   28.728107] input: HDA NVidia Front Headphone as /devices/pci0000:00/0000:00:08.0/sound/card0/input12
[   30.683618] r8169 0000:04:00.0 eth0: link up
[   30.683631] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   30.687857] 8021q: 802.1Q VLAN Support v1.8
[  230.231736] perf: interrupt took too long (2511 > 2500), lowering kernel.perf_event_max_sample_rate to 79000
[  440.135449] retroarch[552]: segfault at c18 ip 00000000004d7810 sp 00007f349dd0eb78 error 4 in retroarch[400000+29d000]

ToKe79, 2018-06-07:
Hi,
could you please provide output of following commands to see what is mounted etc.:
```
blkid
mount
df
```

TeamKinetic, 2018-06-17:
Hi there,

Thanks for the response, here is the requested output;
blkid
/dev/sda1: LABEL="System" UUID="c78975e3-102c-4a1d-bf99-b25be26afade" TYPE="ext4" PARTUUID="e27658c6-01"
/dev/sda2: LABEL="Storage" UUID="922f87bc-c296-4fd3-8d04-c79dad57d446" TYPE="ext4" PARTUUID="e27658c6-02"
/dev/sdb1: SEC_TYPE="msdos" LABEL="LAKKA" UUID="2111-3926" TYPE="vfat" PARTLABEL="system" PARTUUID="41fa678d-0d1b-4b55-a24a-721867915ed5"
/dev/sdb2: LABEL="LAKKA_DISK" UUID="b9ef8ad9-785d-4546-a992-032454f63ce7" TYPE="ext4" PARTLABEL="primary" PARTUUID="a4dd3e7b-469e-4208-a354-915748d987e7"
/dev/loop0: TYPE="squashfs"

mount
devtmpfs on /dev type devtmpfs (rw,relatime,size=879404k,nr_inodes=219851,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/sdb1 on /flash type vfat (ro,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdb2 on /storage type ext4 (rw,noatime,data=ordered)
/dev/loop0 on / type squashfs (ro,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/usr/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
tmpfs on /storage/.cache/machine-id type tmpfs (ro,mode=755)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
tmpfs on /var type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev)
none on /tmp/database type overlay (rw,relatime,lowerdir=/usr/share/libretro-database,upperdir=/storage/database,workdir=/storage/.tmp/database-workdir)
none on /tmp/shaders type overlay (rw,relatime,lowerdir=/usr/share/common-shaders,upperdir=/storage/shaders,workdir=/storage/.tmp/shaders-workdir)
none on /tmp/assets type overlay (rw,relatime,lowerdir=/usr/share/retroarch-assets,upperdir=/storage/assets,workdir=/storage/.tmp/assets-workdir)
none on /tmp/cores type overlay (rw,relatime,lowerdir=/usr/lib/libretro,upperdir=/storage/cores,workdir=/storage/.tmp/cores-workdir)
none on /tmp/joypads type overlay (rw,relatime,lowerdir=/etc/retroarch-joypad-autoconfig,upperdir=/storage/joypads,workdir=/storage/.tmp/joypads-workdir)

df
Filesystem           1K-blocks      Used Available Use% Mounted on
devtmpfs                879404    317796    561608  36% /dev
/dev/sdb1               524008    331136    192872  63% /flash
/dev/sdb2               482664    478568         0 100% /storage
/dev/loop0              317824    317824         0 100% /
tmpfs                   882556         0    882556   0% /dev/shm
tmpfs                   882556      6856    875700   1% /run
tmpfs                   882556         0    882556   0% /sys/fs/cgroup
tmpfs                   882556      6856    875700   1% /storage/.cache/machine-id
tmpfs                   882556        20    882536   0% /var
tmpfs                   882556         0    882556   0% /tmp
none                    482664    478568         0 100% /tmp/database
none                    482664    478568         0 100% /tmp/shaders
none                    482664    478568         0 100% /tmp/assets
none                    482664    478568         0 100% /tmp/cores
none                    482664    478568         0 100% /tmp/joypads

ToKe79, 2018-06-21:
the /storage mount point is full, which is odd - during the first boot the second partition on the "Lakka" storage device should be expanded to the remaining capacity... but according to dmesg it is a 1GB thumb drive and that adds up (512M FAT partition + 471M ext4 partition).

The first boot you mention - is it fresh flashed thumb drive? Because during first boot it should not boot to RetroArch. So do you mean the boot after the formatting of the second partition on the thumb drive is done work everything as should and you have your internal drive partitions mounted in /storage/roms/xxx ?

@kivutar @gouchi could it be that due no free space on the /dev/sdb2 (/storage) it would not mount other drives?

@TeamKinetic can you free up some space on the /storage mount point and try again?

TeamKinetic, 2018-06-21:
Hi there,

Thanks for the response.

If a recreate the USB fresh and 'install' for the first time it mounts the local drive no bother as /storage/roms/xxx. As soon as I shut down and reboot the local storage is not mounted.

How can I free up space on the /storage mount point..? I havent put anything on the USB stick apart from the lakka image and then booted it..?

Thanks

ToKe79, 2018-06-21:
can you check the folder sizes on the /storage partition?
```cd /storage ; du -sh * .cache .config .ssh .tmp```

TeamKinetic, 2018-06-21:
Lakka (official): 2.1 (Generic.x86_64)
Lakka:~ # cd /storage
Lakka:~ # du -sh * .cache .config .ssh .tmp
1.0K    assets
1.0K    backup
1.0K    cores
1.0K    database
1.0K    joypads
0       log.txt
12.0K   lost+found
4.0K    playlists
1.0K    remappings
0       reserved
2.0K    roms
1.0K    savefiles
1.0K    savestates
1.0K    screenshots
1.0K    shaders
1.5M    system
1.0K    thumbnails
464.3M  .cache
141.0K  .config
1.0K    .ssh
11.0K   .tmp

ToKe79, 2018-06-21:
`.cache` takes the most space - haven't you tried to download / update something like thumbnails? can you check the item sizes in that folder? (`cd /storage/.cache ; ls -la ; du -sh *`)

TeamKinetic, 2018-06-21:
Lakka:~ # cd /storage/.cache ; ls -la ; du -sh *
total 22
drwxr-xr-x    9 root     root          1024 May 26 22:01 .
drwxr-xr-x   22 root     root          1024 Jun 17 11:27 ..
drwxr-xr-x    2 root     root          1024 May 26 22:01 bluetooth
drwxr-xr-x    3 root     root          1024 May 26 22:01 connman
drwxr-xr-x    2 root     root          9216 May 28 20:16 cores
drwxr-xr-x    3 root     root          1024 May 26 22:01 cron
-r--r--r--    1 root     root            33 Jun 21 09:30 machine-id
drwxr-xr-x   51 root     root          1024 May 26 23:27 mesa
drwxr-xr-x    2 root     root          1024 May 26 22:04 services
drwxr-xr-x    2 root     root          1024 May 26 22:04 ssh
1.0K    bluetooth
8.0K    connman
464.0M  cores
2.0K    cron
4.0K    machine-id
229.0K  mesa
1.0K    services
10.0K   ssh

TeamKinetic, 2018-06-21:
taken up with cores but I havent tried to download anything

TeamKinetic, 2018-06-21:
Loads of files in the core directory like this;
core.!usr!bin!retroarch.1527373730.667

many of them zero bytes

ToKe79, 2018-06-21:
I don't know - can you observe after the reboot, if the /storage partition is filled up again? did you see your drives mounted in /storage/roms after the reboot?

ToKe79, 2018-06-22:
It looks like to core dumps are stored there, when retroarch segfaults. You can remove any core dumps at every boot by adding:
```
rm -rf /storage/.cache/cores/*
```
to `/storage/config/autostart.sh`. If you do not have autostart.sh, you can create it using nano (`nano /storage/.config/autostart.sh`. First line must be `#!/bin/sh`. To save the file, press Ctrl+X in nano and confirm. Add execute permisson to the file too (`chmod +x /storage/.config/autostart.sh`).

GeorG323, 2018-11-18:
Thank you guys, I created an account just to let you guys know that this tread also help me, Lakka wasn't mounting my PC internal storage and it was also due to my usb being full.

closed 2019-06-26 by ToKe79