libretro/Lakka-LibreELEC #479: Saitek P880 not recognized on Odroid C1

stiegenhaus, 2018-05-29:
My Saitek P880 on Odroid C1 (Lakka-OdroidC1.arm-2.1-devel-20180529094545-r28190) doesn't  show on the screen/system does not react to any input of it.

dmesg:

[   34.486848@3] usb 1-1.3.4: new low-speed USB device number 9 using dwc_otg
[   39.609661@3] usbhid 1-1.3.4:1.0: can't add hid device: -32
[   39.609692@3] usbhid: probe of 1-1.3.4:1.0 failed with error -32

lsusb:
Bus 001 Device 010: ID 06a3:0109 Saitek PLC P880 Pad

On a rpi3b, recent nightly, no issues with the same controller.

For comparison with my Saitek P990:

dmesg:
[  128.345095@3] usb 1-1.3.4: new low-speed USB device number 9 using dwc_otg
[  128.476365@3] input: Saitek P990 Dual Analog Pad as /devices/lm1/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.0/input/input2
[  128.476907@3] hid-generic 0003:06A3:040B.0002: input,hidraw0: USB HID v1.00 Joystick [Saitek P990 Dual Analog Pad] on usb-lm1-1.3.4/input0

lsusb:
Bus 001 Device 009: ID 06a3:040b Saitek PLC P990 Dual Analog Pad

Someone any idea?

ToKe79, 2018-05-29:
Hi,
most probably module is not loaded. Try `modprobe hid_saitek`. According to linux configfile for Odroid driver should be compiled as module.

stiegenhaus, 2018-05-30:
Thanks for the fast reply. I tried `modprobe hid_saitek` with no change. Same dmesg output (can't add hid device).
Strange thing here is, that the P990 works without any issue but the P880 doesn't.

stiegenhaus, 2019-11-19:
Changed my hardware to RPi4 and no issues with P880 anymore on 2.3.1 build

closed 2020-08-18 by Ntemis