libretro/Lakka-LibreELEC #482: Installation on Raspberry Zero W a complete disaster

southbot, 2018-06-02:
Downloaded from the Lakka site. Installed and all seemed to be OK.
Does not detect my gamepad even though other progs such as RetroPie detect it.
I used a USB keyboard to do WIFI setup.  It scans and finds my wifi signal but when I select it it just hangs.
It seems to be expecting a gamepad input because it shows an on screen keyboard.
I tried downloading from Git but am not sure how to install it because etcher says it is an invalid image.
I tried setting up wifi in config file but that did not help.

RobLoach, 2018-06-02:
Thanks for testing this out.

> Does not detect my gamepad even though other progs such as RetroPie detect it.

Which controller? Would be great to confirm a config file in https://github.com/libretro/retroarch-joypad-autoconfig

> I tried downloading from Git but am not sure how to install it because etcher says it is an invalid image.

Downloading what from git?

> It seems to be expecting a gamepad input because it shows an on screen keyboard.

You should be able to still use your USB keyboard when the on-screen display comes up.

southbot, 2018-06-02:
> display comes up. When it gets to this screen the USB keyboard is
> dead. I tried every key. Nothing works.

Hi

Sorry I ranted a bit. It was just very frustrating and time wasting.

RobLoach, 2018-06-03:
No worries, when things don't function as expected, it can be frustrating. If you find the log files, it'll tell you what vendor and product ID the controller is. Then we can have it added to the autoconf.

southbot, 2018-06-03:
Unfortunately I overwrote that build so I don’t have the log file.
I am about to test a nightly build and if I have the same problems I will post the log file.

southbot, 2018-06-03:
Where do I find the correct log file for this info?

southbot, 2018-06-03:
Ok this is what I got from win7 PC for controller. It is a PS2 compatible device

HID\VID_0C12&PID_0005&REV_0001
HID\VID_0C12&PID_0005
HID_DEVICE_SYSTEM_GAME
HID_DEVICE_UP:0001_U:0004
HID_DEVICE

ToKe79, 2018-06-04:
you can use the keyboard to navigate the RetroArch XMB menu and create own joypad configuration file / button binding. but look through dmesg if the controller is correctly detected, i.e. leave the controller unplugged during boot, connect via ssh, execute `dmesg -c > /storage/dmesg1.log` (this will store the output to file and clear the log), plug in the controller, execute `dmesg > /storage/dmesg2.log` (this will store new logged messages in different file). In case you are unable to bind your controller, post the logs here. You can use e.g. WinSCP to copy the log files from your Lakka device.

southbot, 2018-06-04:
I have attached the log files.
I currently dont have a screen attached but yester I tried binding but it does not detect any activity from the gamepad.
[dmesg1.log](https://github.com/libretro/Lakka-LibreELEC/files/2067472/dmesg1.log)
[dmesg2.log](https://github.com/libretro/Lakka-LibreELEC/files/2067473/dmesg2.log)

southbot, 2018-06-04:
oops pressed wrong button :(

southbot, 2018-06-04:
I there a reason why the gamepad controller config files have to be read only? Is there no way to put them in a directory that is easy to write to?

ToKe79, 2018-06-04:
the lakka root-fs is read only (as it is only an image using squashfs). however there is some kind of "overlay" feature, i.e. the read only system folder `/etc/retroarch-joypad-autoconfig` is mounted as `/tmp/joypads`, but files stored in `/storage/joypads` overrule these files, so you can put your config file to e.g. `/storage/joypads/udev`.

gouchi, 2018-06-04:
The device 0c12:0005 required [CONFIG_HID_ZEROPLUS](https://cateee.net/lkddb/web-lkddb/HID_ZEROPLUS.html) to be enabled which is not set for RPi platform.

southbot, 2018-06-04:
So as a linux noob am I to assume this will require the kernel to be re-compiled?
Why is this required? Is a PS2 gamepad not a very common thing?

gouchi, 2018-06-04:
Yes.

It is enabled for Generic platform but we don't enabled it for every platform more depending on the user's case.

ToKe79, 2018-06-04:
http://nightly.builds.lakka.tv/special_builds/rpi_zeroplus/
you can test these images

southbot, 2018-06-04:
I tried putting a config file in storage/jpypads/udev but did not work.
I will try the new image as soon as I have downloaded it

southbot, 2018-06-04:
I used Lakka-RPi2.arm-2.1-devel-20180604232239-r28190-gcc135d4.img.gz
It did not boot at all

ToKe79, 2018-06-04:
I am not very familiar with the raspis, but try the other image. I built all rpi targets - if you boot previously image with RPi in the file name (not RPi2) then use RPi.arm file.

southbot, 2018-06-04:
lol - Yes I deleted that post. As I posted it I realised that I loaded the wrong one :)
The game pad is working now. Thank you very much for your help. 
I have not tested it with a game yet, but it is able to control the menu. I will try it with a game and see how it works

ToKe79, 2018-06-05:
Let us know your results. Thank you.

southbot, 2018-06-05:
It seems to be working fine. I cannot comment about the vibration part because I dont know which roms would use that if any.
The only problem I had was after re-mapping the autoconfig save said that it could not save the data. The save did work later on a second attempt.
Thanks again for the help

ToKe79, 2018-06-05:
Thank you for your feedback and enjoy your games.

PR #484 (2018-06-18): RPi/RPi2: add support for Zeroplus based gamepads
This adds support for Zeroplus based gamepads.
Fixes #482
closed 2018-06-05 by ToKe79