libretro/Lakka-LibreELEC #486: Lakka's Flower flashing in loop after installing

Phelxl, 2018-06-17:
Hi, it's two days that i try to install lakka on my pc.When i boot the first time i select the installer all goes fine, then at the end of the install it says to remove the usb/cd/dvd of lakka and reboot, i do so but when i boot up the second time, during lakka's gui boot, i get the logo flashing every 2 seconds infinitely. 

-Phelxl

Phelxl, 2018-06-17:
Here are the logs:
[dmesg.txt](https://pastebin.com/rYfLB89r)
[lsusb.txt](https://pastebin.com/WH9XR9zp)
[retroarch.txt](https://pastebin.com/nJWLBXFu)
[lscpi.txt](https://pastebin.com/ubFuG67u)

gouchi, 2018-06-17:
Thank for the logs.

[Same issue](https://github.com/libretro/Lakka-LibreELEC/issues/312), can you try to launch RA with
`DRAW_USE_LLVM=0 retroarch -v`

Phelxl, 2018-06-20:
I have to boot in textmode and then use that command? If so i tried, still logo flashing

gouchi, 2018-06-20:
Try to boot in text mode by entering in the menu prompt 
`live textmode`

Then launch RA with
` DRAW_USE_LLVM=0 retroarch -v`

If you need to enable ssh please boot with
`live textmode ssh`

Thank you.

mrc00l88, 2019-01-06:
I had this same issue - I was able to boot with live textmode then launch RA with DRAW_USE_LLVM=0 retroarch -v... 

Now, what do I do from here? What file(s) do I need to edit to be able to install Lakka to the HDD, and have it boot into Lakka like normal?

gouchi, 2019-01-21:
@mrc00l88 Install Lakka normally by entering `installer` in the [boot prompt](http://www.lakka.tv/get/linux/generic/install/first-boot/) then edit [autostart.sh](https://wiki.libreelec.tv/autostart.sh) and [stop RA](http://www.lakka.tv/doc/Troubleshooting-Lakka/) then launch it with `DRAW_USE_LLVM=0 retroarch` so you should have something

`systemctl stop retroarch.service && DRAW_USE_LLVM=0 retroarch`

mrc00l88, 2019-01-21:
@gouchi thank you. This did NOT work for me with Lakka 2.2 release, but it DID work in Lakka  Generic.x86_64-3.0-devel-20180617091802-87598f2.img.gz

However, once in Lakka 3.0, I could NOT enter a Wifi password (key presses were doubled) - is there a way to manually set the Wifi password? Seems that's the last hurdle for me in using Lakka again on this new PC.

mrc00l88, 2019-01-21:
Cant seem to edit my post... the 3.0 version I got in to was Lakka-Generic.x86_64-3.0-devel-20180703104709-2fb13c2.img.gz   - I posted the incorrect 'June' version above as I was navigating to redownload 3.0 after the 2.2 version fails to run at all.

gouchi, 2019-01-23:
@mrc00l88 is it possible to get [graphic card log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#graphic-card-logs) ? We may need a newer kernel to support your intel gpu properly.

ToKe79, 2019-07-02:
no response

closed 2019-07-02 by ToKe79