libretro/Lakka-LibreELEC #487: 60fps smooth launcher screen?

juj, 2018-06-17:
I was testing today my https://github.com/juj/fbcp-ili9341 project to run on Lakka, latest version downloaded today. Seems to work nicely with fbcp-ili9341 without any changes needed.

[image]
[image]

Running [fbcp-ili9341s frame delivery statistics histogram](https://youtu.be/IqzKT33Rwjc) profiling tool, I get results that look like the above. The profiler shows there is quite a bit of microstuttering present that seems to be coming from the rendering of the text and icons, since the stuttering is very dependant on what menu elements exist on screen.

In the first image there is a minimal amount of text and icons on screen, which yields an average fps of ~55. Moving the menu two elements to the left drops fps to average 45, i.e. a reduction from 18.12msecs/frame to 22.22msecs/frame -> a +4.1msecs more time spent on average. Looking at the graph, the menu is not GPU rendering/fillrate bottlenecked, since the Pi 3B is able to produce several consecutive frames at within 16.66msecs, but the more content there is on the screen, the more intermittent stuttering there is present.

I should say I don't currently use Lakka, so this is not a problem for me or anything. Just thought this might be interesting data for someone, as this looks like there is some headroom to optimize to get a more smoother 60fps launcher menu. The project looks great, keep up the amazing work!

juj, 2018-06-18:
Hmm, something that I then notice is that when booting on a Pi Zero instead of Pi 3B, it does give smooth 60fps:

[image]

oddly Pi 3B runs the BCM core at 250MHz in the launcher without turboing up, whereas Pi Zero sits at a constant turbo up at 400MHz. With Pi Zero, I did not have a HDMI cable connected, giving

```
Lakka (official): 2.1 (RPi.arm)
Lakka:~ # tvservice -s
state 0x40001 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
```

Ntemis, 2018-06-18:
you can find out here more details https://github.com/juj/fbcp-ili9341/blob/master/README.md
closing down

closed 2018-06-18 by Ntemis