libretro/Lakka-LibreELEC #488: Lakka wont compile due to missing repository

alphanu1, 2018-06-18:
compile error as follows:

  UNPACK   higan-sfc-balanced
Cloning into '/mnt/8c026599-057d-4201-8616-ab6cbff6593e/lakka_X11/build.Lakka-Generic.x86_64-2.1-devel/higan-sfc-balanced-f43168a'...
Username for 'https://github.com': alphanu1
Password for 'https://alphanu1@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/hex-usr/nSide/' not found
Makefile:27: recipe for target 'image' failed
make: *** [image] Error 128

RobLoach, 2018-06-18:
I believe nSide was deprecated by the author. Made an issue in libretro-super for this too https://github.com/libretro/libretro-super/issues/824

closed 2018-06-18 by kivutar

alphanu1, 2018-06-18:
There are still compile issues here, its now just after cloning the new gti.

  UNPACK   higan-sfc-balanced
Cloning into '/mnt/8c026599-057d-4201-8616-ab6cbff6593e/lakka_X11/build.Lakka-Generic.x86_64-2.1-devel/higan-sfc-balanced-f43168a'...
remote: Counting objects: 2506, done.
remote: Compressing objects: 100% (1968/1968), done.
remote: Total 2506 (delta 496), reused 2506 (delta 496), pack-reused 0
Receiving objects: 100% (2506/2506), 2.52 MiB | 572.00 KiB/s, done.
Resolving deltas: 100% (496/496), done.
Checking connectivity... done.
error: pathspec 'f43168a' did not match any file(s) known to git.
Makefile:27: recipe for target 'image' failed
make: *** [image] Error 1

@RobLoach 
@kivutar

RobLoach, 2018-06-18:
https://github.com/libretro/Lakka-LibreELEC/pull/491 :+1:  Thanks

PR #489 (2018-06-18): higan-sfc-balanced: Update nSide git URL
Fixes #488
commit e5257810 (2018-06-18): higan-sfc-balanced: Update nSide git URL (#489)

Fixes #488