libretro/Lakka-LibreELEC #492: OverlayFS mount problem with network boot

smoothquark, 2018-06-19:
Apologies if the solution is obvious. I am trying to boot Lakka via PXE as I do LibreELEC. My system is diskless. Boots fine but icons are just black. Looks like the overlayFS mounts are not being mounted.

PXE config:
DEFAULT Lakka
PROMPT 0

LABEL Lakka
	KERNEL boot.Lakka/KERNEL.Generic.x86_64
	APPEND ip=dhcp ssh boot=NFS=192.168.1.145:/srv/nfs/LibreELEC/Lakka disk=NFS=192.168.1.145:/srv/nfs/LibreELEC/Storage/@UID@

relevant part on dmesg:
[   12.301540] overlayfs: filesystem on '/storage/shaders' not supported as upperdir
[   12.302005] overlayfs: filesystem on '/storage/database' not supported as upperdir
[   12.303170] overlayfs: filesystem on '/storage/cores' not supported as upperdir
[   12.303414] overlayfs: filesystem on '/storage/assets' not supported as upperdir
[   12.306119] overlayfs: filesystem on '/storage/joypads' not supported as upperdir

mount:
devtmpfs on /dev type devtmpfs (rw,relatime,size=1856640k,nr_inodes=464160,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
192.168.1.145:/srv/nfs/LibreELEC/Lakka on /flash type nfs (ro,noatime,vers=3,rsize=32768,wsize=32768,namlen=255,soft,nolock,proto=tcp,port=2049,timeo=3,retrans=2,sec=sys,local_lock=all,addr=192.168.1.145)
192.168.1.145:/srv/nfs/LibreELEC/Storage/7085c238809e on /storage type nfs (rw,noatime,vers=3,rsize=32768,wsize=32768,namlen=255,soft,nolock,proto=tcp,port=2049,timeo=3,retrans=2,sec=sys,local_lock=all,addr=192.168.1.145)
/dev/loop0 on / type squashfs (ro,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/usr/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
mqueue on /dev/mqueue type mqueue (rw,relatime)
tmpfs on /var type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)

Could this be because /storage itself is an NFS mount?

Thank you.

smoothquark, 2018-06-20:
Forgot to add - I can get the icons, fonts etc to come up if I do not use disk=NFS=192.168.1.145:/srv/nfs/LibreELEC/Storage/@uid@ in the pxe config, but this means that I have no way of saving my settings as there is no more persistent storage

ToKe79, 2018-06-20:
Hi, I guess it is not supported by the kernel we use (https://www.phoronix.com/scan.php?page=news_item&px=OverlayFS-NFS-Export-Linux-4.16) - Lakka uses 4.11 for `Generic` builds. Maybe there is a workaround (using `mount -o bind /what /where` and then mounting the upper/lower dir with overlayfs, but I did not test this).

smoothquark, 2018-06-20:
Thanks. It looks like NFS is indeed not supported as an upper file system (https://www.kernel.org/doc/Documentation/filesystems/overlayfs.txt) - 4th paragraph under "Overlay objects". Shall have a play to see if mount bind or an alternative works.

kivutar, 2018-06-24:
We should consider removing the OverlayFS stuff. Maybe it adds too much complexity compared to the functionality. We use it mainly to allow people overriding assets and joypad configs. And it comes at a maintenance cost.

ToKe79, 2018-06-24:
Anyway, as a solution, you could just point your the RetroArch dirs to /storage/.config/retroarch/<assets|thumbnails|database|...> and move the files to these folders.

Here is what I would do (hopefully there are no typos):
```
# stop RetroArch
systemctl stop retroarch

# copy files from read only file system to the storage
cp -r /usr/lib/libretro/* /storage/.config/retroarch/cores/
cp -r /usr/share/libretro-database/* /storage/.config/retroarch/database/
cp -r /usr/share/retroarch-assets/* /storage/.config/retroarch/assets/
cp -r /etc/retroarch-joypad-autoconfig/* /storage/.config/retroarch/joypads/
cp -r /usr/share/common-shaders/* /storage/.config/retroarch/shaders/

# replace the paths in the RetroArch config file
sed -e "s|/tmp/|/storage/.config/retroarch/|g" -i /storage/.config/retroarch/retroarch.cfg
```

yangxuan8282, 2018-06-25:
got similar issues on rpi3, although not use overlayfs, just nfs
and seems @ToKe79 's solution solve the problems 

use network boot rpi with LibreELEC everything work fine

ToKe79, 2018-06-25:
Closing this issue, as the feature is not supported (using NFS as upper file system). @kivutar I think the overlayfs is a good feature for Lakka. However, without it the users still have the possibility to replace/add assets/joypads/... but needs to tweak Lakka as suggested above. The removal of overlayfs feature is not the point of this thread :-)

closed 2018-06-25 by ToKe79