libretro/Lakka-LibreELEC #499: Playstation Portable (PPSSPP) not working (x64)

masid101, 2018-07-07:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 2.1.1
- What system hardware are you using? PC x64
- What did you do? Run a Playstation Portable game with PPSSPP
- What did you expect to happen? The game to work
- What happened instead? PPSSPP crashed (worked with Lakka 2.1)

Here's the log ...

[INFO] RetroArch 1.7.3 (Git 2a2e732)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES 
Built: Jun 23 2018
[INFO] Version: 1.7.3
[INFO] Git: 2a2e732
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 5 connectors.
[INFO] [DRM]: Connector 0 connected: no
[INFO] [DRM]: Connector 0 has 0 modes.
[INFO] [DRM]: Connector 1 connected: yes
[INFO] [DRM]: Connector 1 has 33 modes.
[INFO] [DRM]: Connector 1 assigned to monitor index: #1.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Connector 3 connected: yes
[INFO] [DRM]: Connector 3 has 34 modes.
[INFO] [DRM]: Connector 3 assigned to monitor index: #2.
[INFO] [DRM]: Connector 4 connected: no
[INFO] [DRM]: Connector 4 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 4: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 8: (1600x1200) 1600 x 1200, 60 Hz
[INFO] [DRM]: Mode 9: (1400x1050) 1400 x 1050, 60 Hz
[INFO] [DRM]: Mode 10: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 11: (1360x768) 1360 x 768, 60 Hz
[INFO] [DRM]: Mode 12: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 13: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 14: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 15: (1024x768) 1024 x 768, 75 Hz
[INFO] [DRM]: Mode 16: (1024x768) 1024 x 768, 70 Hz
[INFO] [DRM]: Mode 17: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 18: (800x600) 800 x 600, 75 Hz
[INFO] [DRM]: Mode 19: (800x600) 800 x 600, 72 Hz
[INFO] [DRM]: Mode 20: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 21: (800x600) 800 x 600, 56 Hz
[INFO] [DRM]: Mode 22: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 23: (720x576i) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 24: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 25: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 26: (720x480i) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 27: (720x480i) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 28: (640x480) 640 x 480, 75 Hz
[INFO] [DRM]: Mode 29: (640x480) 640 x 480, 73 Hz
[INFO] [DRM]: Mode 30: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 31: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 32: (720x400) 720 x 400, 70 Hz
[... 300 lines cut ...]
[INFO] [GLSL]: Linking GLSL program.
[INFO] Resetting shader to defaults ... 
[INFO] [GL]: Using 1 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Initializing HW render (512 x 512).
[INFO] [GL]: Max texture size: 16384 px, renderbuffer size: 16384 px.
[INFO] [GL]: Supports FBO (render-to-texture).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event4).
[INFO] [udev]: Mouse #0 (/dev/input/event6).
[INFO] [udev]: Mouse #1 (/dev/input/mouse0).
[INFO] [udev]: Plugged pad: PLAYSTATION(R)3 Controller (1356:616) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event15) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event15) supports 16 force feedback effects.
[INFO] [Autoconf]: 208 profiles found.
[INFO] [autoconf]: selected configuration: /tmp/joypads/udev/Sony-PlayStation3-DualShock3-Controller-Bluetooth.cfg
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[libretro INFO] [G3D] Context reset
Segmentation fault (core dumped)

brandleesee, 2018-07-11:
@kivutar we spoke about the PPSSPP logs: [built-in](https://raw.githubusercontent.com/brandleesee/FAQ/master/backups/RetroArch/Lakka/Logs/11Jul18/built-in_log.txt) | [latest in storage/cores](https://raw.githubusercontent.com/brandleesee/FAQ/master/backups/RetroArch/Lakka/Logs/11Jul18/ppsspp_latest_2807100035_log.txt) | [latest in .config/retroarch](https://raw.githubusercontent.com/brandleesee/FAQ/master/backups/RetroArch/Lakka/Logs/11Jul18/latest-core-in-config--retroarch_log.txt)

Also, cache is not empty ([full file](https://raw.githubusercontent.com/brandleesee/FAQ/master/backups/RetroArch/Lakka/Logs/11Jul18/LAKKA_DISK.txt))

```
/media/user/LAKKA_DISK/.cache/cores:
.
..
core.!usr!bin!retroarch.1530603449.378
core.!usr!bin!retroarch.1530806598.602
core.!usr!bin!retroarch.1530808144.367
core.!usr!bin!retroarch.1530813409.367
core.!usr!bin!retroarch.1530813497.595
core.!usr!bin!retroarch.1530813584.605
core.!usr!bin!retroarch.1530813750.615
core.!usr!bin!retroarch.1530873386.369
core.!usr!bin!retroarch.1530944578.369
core.!usr!bin!retroarch.1530950250.368
```

Reference in forums is [here](https://forums.libretro.com/t/lakka-v2-1-1-silently-released-2018-06-25-info-and-discussion/16797/13?u=yrvyne).

Thank you.

ShockwaveTheFallen, 2018-09-30:
@kivutar @gouchi @Ntemis Tried sliding the PPSSPP core into the Cores menu via SSH to try to remedy the situation, didn't work either. Segmentation fault just the same, and it's crashing all the games with the same log above. Just wanted to give you guys a heads up. We all have lives, and I'm right there with you on the whole "busy as hell" front.

ShockwaveTheFallen, 2018-09-30:
@gouchi Sorry bud, crashed again with the latest nightly. Same error, failed to load the core.

ShockwaveTheFallen, 2018-10-01:
`login as: root
root@192.168.0.12's password:
Access denied
root@192.168.0.12's password:
Access denied
root@192.168.0.12's password:
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (official): devel-20180914000711-r28217-g74bf6c6 (Generic.x86_64)
Lakka:~ # stop retroarch
-sh: stop: not found
Lakka:~ # retroarch=0
Lakka:~ # systemctl stop retroarch
Lakka:~ # retroarch -v
[INFO] RetroArch 1.7.4 (Git 1155e11)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES
Built: Sep  9 2018
[INFO] Version: 1.7.4
[INFO] Git: 1155e11
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [SoftFilter]: Found plug: 2xBR (2xbr).
[INFO] [SoftFilter]: Found plug: 2xSaI (2xsai).
[INFO] [SoftFilter]: Found plug: Blargg NTSC SNES (blargg_ntsc_snes).
[INFO] [SoftFilter]: Found plug: Darken (darken).
[INFO] [SoftFilter]: Found plug: EPX (epx).
[INFO] [SoftFilter]: Found plug: LQ2x (lq2x).
[INFO] [SoftFilter]: Found plug: Phosphor2x (phosphor2x).
[INFO] [SoftFilter]: Found plug: Scale2x (scale2x).
[INFO] [SoftFilter]: Found plug: Super2xSaI (super2xsai).
[INFO] [SoftFilter]: Found plug: SuperEagle (supereagle).
[INFO] Using 1 threads for softfilter.
[INFO] [Video]: Video @ fullscreen
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: no
[INFO] [DRM]: Connector 0 has 0 modes.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: yes
[INFO] [DRM]: Connector 2 has 27 modes.
[INFO] [DRM]: Connector 2 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 5: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 8: (1680x1050) 1680 x 1050, 60 Hz
[INFO] [DRM]: Mode 9: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 10: (1440x900) 1440 x 900, 60 Hz
[INFO] [DRM]: Mode 11: (1280x960) 1280 x 960, 60 Hz
[INFO] [DRM]: Mode 12: (1280x800) 1280 x 800, 75 Hz
[INFO] [DRM]: Mode 13: (1280x800) 1280 x 800, 60 Hz
[INFO] [DRM]: Mode 14: (1280x768) 1280 x 768, 60 Hz
[INFO] [DRM]: Mode 15: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 16: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 17: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 18: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 19: (800x600) 800 x 600, 56 Hz
[... 588 lines cut ...]
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[ERROR] Failed to open libretro core: "/tmp/cores/ppsspp_libretro.so"
[ERROR] Error(s): libsnappy.so.1: cannot open shared object file: No such file or directory
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x797fe0
[INFO] [MIDI]: Initializing ...
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "null" driver.
[ERROR] Implementation uses threaded audio. Cannot use rewind..
[WARN] Input device ID 5 is unknown to this libretro implementation. Using RETRO_DEVICE_JOYPAD.
[WARN] Input device ID 5 is unknown to this libretro implementation. Using RETRO_DEVICE_JOYPAD.
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [GL]: VSync => on
[INFO] [GL]: VSync => on
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
`

ShockwaveTheFallen, 2018-10-02:
Nope, failed again.

leranger, 2018-10-06:
Hi,
It seems that latest PPSSPP core needs snappy compression library (https://github.com/google/snappy).

Is it provided with Lakka ?

IgorWalkers, 2018-10-24:
Hey,
How is it going? PSP games still not playable?

gouchi, 2018-10-25:
Indeed I reproduce the segfault with [latest nightly available](http://nightly.builds.lakka.tv/2018-10-25/).

smoky19x, 2018-12-05:
It cost me a few hours but in the end I was able to narrow down the bug :)

In the sources of PPSSPP there is the file "ext/glew/glew.c" Here is the error in the query in line 20875 of the function glewInit():

```
20875: #elif! Defined (__ APPLE__) || defined (GLEW_APPLE_GLX) / * _UNIX * /
20876: return glxewInit ();
20877: #else
20878: return r;
```

The condition evaluates to true, so line 20876 calls "glxewInit()", which is not (and does not have to be) implied in Lakka, which eventually causes a Segmentation Fault elsewhere.

When the lines 20875-20877 are commented out and PPSSPP is recompiled, PPSSPP runs again without problems.

Since I'm still pretty new to Linux development, I do not know how best to proceed here. In my view, the condition on line 20875 is incorrect. However, the glew.c file comes directly from the PPSSPP repository: https://github.com/hrydgard/ppsspp/blob/master/ext/glew/glew.c, which in turn comes from the GLEW project: https://github.com/nigels-com/glew. Most likely, the maintainers should be notified and perhaps a pull request submitted by an experienced developer. But I do not know enough about the details to do this myself. Could that please take over one of the main developers or guide me if necessary?

In the meantime maybe a temporary patch in "/packages/libretro/ppsspp/patches" could be created to make PPSSPP on Lakka run again. A lot of people would be happy about that ;)

https://forums.libretro.com/t/psp-not-working-in-x64/18110
https://forums.libretro.com/t/trouble-getting-any-psp-title-to-run/14432
https://forums.libretro.com/t/segmentation-fault-core-dumped-on-ppsspp-core/16910
https://forums.libretro.com/t/ppsspp-in-stable-2-1-1-not-functioning/17537
https://forums.libretro.com/t/ppsspp-core-from-2-years-ago/19058

commit 05526c3e (2018-12-24): PPSSPP patch. Fixes #499
closed 2018-12-24 by kivutar