libretro/Lakka-LibreELEC #501: Lakka 2.1 Generic x86-64 installer does not work

Kriskras99, 2018-07-10:
I tried installing Lakka on my custom build pc and it wouldn't go past the boot prompt. (Wait for the live installer or press tab). Pressing tab gives me the option to type 'installer' or 'live' and after pressing enter the cursor flashes for a few seconds and then disappears/hangs.
I've also tried using Lakka-Generic.x86_64-2.1-devel-canary-dfa8b67.img and tried both on my laptop (Lenovo ThinkPad P51) but the same things happens.

In the end I had to fall back to Lakka-Generic.x86_64-2.0.img which does work.
I can update it to version 2.1 without a problem, so it's only an issue in the installer.

I can't provide any logs because it fails to pass the bootloader stage.

gouchi, 2018-07-19:
Can you try to burn your image with [Etcher](https://etcher.io/) ?

We have updated syslinux to version 6.04-pre1 maybe that is why.

Kriskras99, 2018-08-04:
It does work with Etcher, so something goes wrong with dd (I'm on linux).
Maybe update the docs for linux to use Etcher instead of dd for now?

gouchi, 2018-08-08:
Thank you for the feedback.

We will try to update the documentation to use Etcher.

closed 2018-08-08 by gouchi