libretro/Lakka-LibreELEC #505: Core dump while booting retroarch

yroffin, 2018-07-20:
Context:

- Version of Lakka 
2.1.1
Linux Lakka 4.11.12-rt14 #1 SMP PREEMPT RT Sun Jun 17 19:40:49 CEST 2018 x86_64 GNU/Linux

- System hardware
 ACER ASPIRE 7520
 GEFOCE 7000M

my issue i core dump in retroarch start (i assume retroarch don't find correctly output video)

```
[INFO] RetroArch 1.7.3 (Git 2a2e732)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3
Built: Jun 23 2018
[INFO] Version: 1.7.3
[INFO] Git: 2a2e732
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 8 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: yes
[INFO] [DRM]: Connector 2 has 8 modes.
[INFO] [DRM]: Connector 2 assigned to monitor index: #2.
[WARN] [DRM]: Couldn't find DRM encoder.
[WARN] [DRM]: Couldn't get device resources.
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
function is no-op
function is no-op
function is no-op
function is no-op
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)

```

```
lspci -nnk | grep -A 3 VGA
00:12.0 VGA compatible controller [0300]: NVIDIA Corporation C67 [GeForce 7000M / nForce 610M] [10de:0533] (rev a2)
        Subsystem: Acer Incorporated [ALI] C67 [GeForce 7000M / nForce 610M] [1025:0126]
        Kernel driver in use: nouveau

```

gouchi, 2018-07-20:
Hi,

Can you try to edit the [configuration file](http://www.lakka.tv/doc/Configuring-Lakka/) and use 
`video_monitor_index = "2"`

Thank you.

yroffin, 2018-07-22:
hi,

is retroarch must setup on top of ubuntu-server ... or xorg is needed ?

gouchi, 2018-07-24:
On Generic platform, we don't use Xorg so we rely on DRM/KMS.

Did you try to make a test after editing the configuration file ?

Thank you.

ToKe79, 2018-09-21:
Hi @gouchi 
using this thread to help a different user with similar configuration (desktop PC, Pentium D 2.8Ghz, 2GB DDR 2 RAM, GeForce 7600GT (256MB)) and same error.

`lspci -nnk` (full output [here](https://vps.vudiq.sk/lakka-issue-505-lspci-mrganchev.txt))
```
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation G73 [GeForce 7600 GT] [10de:0391] (rev a1)
	Kernel driver in use: nouveau
```

`retroarch.log`
```
[INFO] RetroArch 1.7.3 (Git 2a2e732)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Jun 23 2018
[INFO] Version: 1.7.3
[INFO] Git: 2a2e732
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ 960x720
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: no
[INFO] [DRM]: Connector 0 has 0 modes.
[INFO] [DRM]: Connector 1 connected: yes
[INFO] [DRM]: Connector 1 has 17 modes.
[INFO] [DRM]: Connector 1 assigned to monitor index: #1.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[WARN] [DRM]: Couldn't find DRM encoder.
[WARN] [DRM]: Couldn't get device resources.
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
function is no-op
function is no-op
function is no-op
function is no-op
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)
```
`dmesg.log` (`grep -E 'drm|nouveau'`, full log [here](https://vps.vudiq.sk/lakka-issue-505-dmesg-mrganchev.txt))
```
[    0.553005] [drm] radeon kernel modesetting enabled.
[    0.553122] [drm] amdgpu kernel modesetting enabled.
[    0.553743] nouveau 0000:01:00.0: NVIDIA G73 (04b100b1)
[    0.751682] nouveau 0000:01:00.0: bios: version 05.73.22.18.01
[    0.759485] nouveau 0000:01:00.0: fb: 256 MiB GDDR3
[    0.831247] nouveau 0000:01:00.0: DRM: VRAM: 251 MiB
[    0.831250] nouveau 0000:01:00.0: DRM: GART: 512 MiB
[    0.831255] nouveau 0000:01:00.0: DRM: TMDS table version 1.1
[    0.831257] nouveau 0000:01:00.0: DRM: TMDS table script pointers not stubbed
[    0.831260] nouveau 0000:01:00.0: DRM: DCB version 3.0
[    0.831264] nouveau 0000:01:00.0: DRM: DCB outp 00: 01000300 00000028
[    0.831269] nouveau 0000:01:00.0: DRM: DCB outp 01: 03000302 00000000
[    0.831272] nouveau 0000:01:00.0: DRM: DCB outp 02: 04011310 00000028
[    0.831275] nouveau 0000:01:00.0: DRM: DCB outp 03: 04011312 00000000
[    0.831278] nouveau 0000:01:00.0: DRM: DCB outp 04: 020223f1 00c0c080
[    0.831281] nouveau 0000:01:00.0: DRM: DCB conn 00: 1030
[    0.831283] nouveau 0000:01:00.0: DRM: DCB conn 01: 2130
[    0.831286] nouveau 0000:01:00.0: DRM: DCB conn 02: 0210
[    0.831288] nouveau 0000:01:00.0: DRM: DCB conn 03: 0211
[    0.831291] nouveau 0000:01:00.0: DRM: DCB conn 04: 0213
[    0.831762] nouveau 0000:01:00.0: DRM: Saving VGA fonts
[    0.886139] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    0.886140] [drm] Driver supports precise vblank timestamp query.
[    0.887869] nouveau 0000:01:00.0: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
[    0.895558] nouveau 0000:01:00.0: DRM: MM: using M2MF for buffer copies
[    0.895587] nouveau 0000:01:00.0: DRM: Setting dpms mode 3 on TV encoder (output 4)
[    0.960608] nouveau 0000:01:00.0: DRM: allocated 1280x1024 fb: 0x9000, bo f6345000
[    0.961495] fbcon: nouveaufb (fb0) is primary device
[    1.045770] nouveau 0000:01:00.0: DRM: 0xD244: Parsing digital output script table
[    1.165185] nouveau 0000:01:00.0: fb0: nouveaufb frame buffer device
[    1.166323] [drm] Initialized nouveau 1.3.1 20120801 for 0000:01:00.0 on minor 0
```

We copied the `/etc/retroarch.cfg` to `/storage/.config/retroarch/retroarch.cfg` and changed
```
# video_monitor_index = 0
```
to
```
video_monitor_index = 1
```
but the result was same. I think we even had to create the `/storage/.config/retroarch` folder, as it was not present yet. The card has two DVI outputs (tried both with default settings - `# video_monitor_index = 0`, same result), there is on-board ATI X300-based 2D/3D graphics (MoBo: http://www.asrock.com/mb/ATI/775Twins-HDTV/) with 3x RCA output, but we were using the DVI output of the NVidia.

Any clues? Thanks a lot!

gouchi, 2018-09-21:
And what is the output of  ?

`for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done`

Thank you.

necrosaromx, 2018-10-13:
Hi! I'm having same issue with similar Hardware, I'm using an Aspire 5520 that is Turion 64 X2 with a Nvidia GeForce 7000m, I'm having the same issue as @yroffin , and the output of the script @gouchi  asked for in my Laptop is:

```
LVDS-1: connected
TV-1: connected
VGA-1: disconnected 
```

gouchi, 2018-10-13:
@necrosaromx can you try to boot in the boot prompt with` live video=LVDS-1:e video=TV-1:d`. It will disable the TV output. Then can you try to provide some [Lakka system logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/) ?

Thank you.

necrosaromx, 2018-10-17:
Thanks for the response @gouchi , I'm trying to figure how to get the logs as the options `live video=LVDS-1:e video=TV-1:d` give me no output on either the Laptop's screen or the VGA out on an external monitor, maybe there's output on the S-Video output but I need to get a cable... as I have no GUI I cannot use the SSH service as it has no network configuration. I'll post further updates.

gouchi, 2018-10-18:
@necrosaromx no problem you are welcome I forgot to add to enable ssh at boot time
`live video=LVDS-1:e video=TV-1:d ssh`

If you don't  have ethernet try with
`live video=LVDS-1:e video=TV-1:d textmode`

Thank you.

ToKe79, 2019-07-02:
no response

closed 2019-07-02 by ToKe79