libretro/Lakka-LibreELEC #506: make clean command doesn't clear .ccache

natinusala, 2018-07-21:
The make clean command should delete the `.ccache` folder inside of the build folder. 

I had this scenario where a previously aborted build yielded an incomplete .so file, and LD didn't want it, making subsequent builds fail, even after a clean. Deleting `.ccache` solved the issue.

ToKe79, 2018-07-22:
use `make distclean` for wiping the build folders completely.

natinusala, 2018-07-22:
I see, thanks ! What is the point of using clean then ?

natinusala, 2018-07-23:
Shouldn't ccache be considered as a build folder ?

kivutar, 2018-07-26:
I don't know ccache enough

kivutar, 2018-07-27:
The best would be to ask LibreELEC why they designed the build system this way. From what I understand, ccache is detecting when the same compilation is run a second time. They use hashes to make sure that if a file changed, it will be recompiled by gcc.

natinusala, 2018-07-27:
Alright, I'm fine with using distclean in the meantime. There must be a reason why clean doesn't delete it. I'll see what I can find about this.

closed 2018-07-27 by natinusala