libretro/Lakka-LibreELEC #507: Boot loop when first installing Lakka to an old PC

ZachySwag, 2018-07-27:
Hey. I was trying to turn this old PC I got a hold of, but had a problem when booting up. Did some research and managed to get some info on it, but I still don't know how to fix it. It's a segmentation error.

Here's some info:
- Which version of Lakka are you using? **Lakka-Generic.i386-2.1**
- What system hardware are you using? **A old PC (From around 2009)**
- What did you do? **Tried to install Lakka to the PC.**
- What did you expect to happen? **Boot fine I guess**
- What happened instead? **When I booted the OS up it kept flashing and rebooting in a boot loop. Yes I'm 99.9% sure it's not corruption because I redownloaded and flashed it to different flash drives about 3 times.**

This is the retroarch.txt log I got from the the textmode
[retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/2234110/retroarch.txt)

Thanks for any help at all! :)

gouchi, 2018-08-01:
Hi,

Can you try to [edit your configuration file](http://www.lakka.tv/doc/Configuring-Lakka) with `menu_shader_pipeline = "0"`. If it is not in the configuration file please add it.

Please boot with boot arguments `ssh tty retroarch=0`

Thank you.

ToKe79, 2019-07-02:
no response

closed 2019-07-02 by ToKe79