libretro/Lakka-LibreELEC #508: Performance issues in the lakka OS interface and crash or slowdown in some emulators.

Marley2018, 2018-07-29:
Hello, I would like to inform you of some problems that I encounter in my tv box or Android player box box brand X96 with amilogic processor s905x 1.51 GHZ with 2gb ram and 16 rom, in which they are slowness in the Xmb interface of Lakka OS 2.1.1 and also some slowndown in the emulators of Snes9x 1.56.1 with the game Yoshi island after the entrance of the first pipe of the first phase of the first world, and another emulator that is presenting small choke in the sound is the Mgba 0.7 wip with the game Iridion 3d and the worst of all of the emulators is the n64 paralell with the Mario Kart 64 game with slowndown in the house of up to 10 fps with GFX paralell plugin but I switching to the Rice wheel plugin much better then I wonder if this is for lack of optimization of the Lakka OS or Emulators or lack of hardware of my box player, in my opinion I think it is optimization problems, and also compare the performance of these emulators with the same versions of Nintendo Wii Retroarch, and that is strange CPS3 run from good that in logic is heavier than Snes9x and Mgba, so check and maybe correct these possible problems.

jayare5, 2018-09-18:
I use Raspberry Pi 3b, and I can confirm a massive downgrade in performance on the XMB as well.
(Lakka from 2016 worked 60fps smooth on XMB unless you loaded shaders and added shadows, newest Lakka is poor without any of that, and even worse with those).

I did also notice perfomance issues on some emulators, for example Gambatte will skip at least 1 frame every few seconds.

fmaida, 2019-03-16:
Is it only an impression, or Lakka 2.2.2 and the cores are slower?
On my Raspberry Pi 3b+ I'm getting an average of 11 fps while navigating through the menus with the ozone interface, and I'm getting an average of 22 fps while emulating CPS-3 games under FinalBurn Alpha. Until 2.2.1 my raspi was able to run CPS-3 games at (almost) full speed.

I've upgraded lakka from 2.2.1 to 2.2.2 by using the built-in function, so in theory my previous settings should have been keeped AFAIK. I can test different setting configurations if needed of course.

closed 2020-08-18 by Ntemis