libretro/Lakka-LibreELEC #512: Videos showing garbled, but have sound ppsspp

dmschwartzii, 2018-07-31:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? Latest nightly
- What system hardware are you using? S912
- What did you do? Opened psp game
- What did you expect to happen? Game to work
- What happened instead? Videos are garbled.

I submitted this on ppsspp git and was told it was Likely linked with the wrong version of ffmpeg

Ntemis, 2018-07-31:
their ffmpeg version is old and needs updating so no its not lakka's fault
we have garbled video in all the boards even the most supported like the raspberry pi's.
 Their latest offer of ffmpeg version for arm socs is v2.7.2
https://github.com/hrydgard/ppsspp-ffmpeg/commits?author=sergiobenrocha2
from 2015 :)
Look at this:
https://github.com/FFmpeg/FFmpeg/releases

dmschwartzii, 2018-07-31:
FFmpeg specifically does not maintain compatibility between releases. Like saying your USB-C plug doesn't seem to charge your Galaxy S3. It's absolutely expected that randomly updating FFmpeg past the version we're compatible with will cause issues.

Small projects with stable APIs, like zlib or xxhash, are different stories. But FFmpeg breaks their API all the time.

-[Unknown]

dmschwartzii, 2018-08-01:
They're saying their not compatible with the version of ffmpeg you use in lakka

dmschwartzii, 2018-08-01:
What I posted is the response I got from them

Ntemis, 2018-08-01:
we are using their ffmpeg(V2.7.2) not the upstream one v4.0.2 all i was pointing out is that theirs needs updating

dmschwartzii, 2018-08-01:
Ok ill let them know

dmschwartzii, 2018-08-02:
we use 3.0.2 on most platforms.

-[Unknown]

So they suggest you try upgrading to atleast 3.0.2 @Ntemis

dmschwartzii, 2018-08-02:
@Unknownbrackets

closed 2020-08-18 by Ntemis