libretro/Lakka-LibreELEC #513: Support for Mayflash Gamecube USB adapters

Mr-Jake, 2018-08-05:
A number of users on the Libretro forums have reported that Mayflash USB adapters do not work with Lakka.

@gouchi asked them to open an issue on GitHub, but they did not or did not know how, so I am opening an issue for them since I also own a Mayflash adapter and would also be interested in support for this. 

[Forum Link](https://forums.libretro.com/t/mayflash-gamecube-to-usb-adapter-supported/14314)

gouchi had suggested enabling [CONFIG_HID_MAYFLASH](https://cateee.net/lkddb/web-lkddb/HID_MAYFLASH.html)

Ntemis, 2018-08-06:
On what device?

Mr-Jake, 2018-08-06:
The forum users reported PC (Generic.x86_64) and RPi3.

It would be better if they would post the details here themselves, but I guess they only post in the Libretro forum and don't want to use GitHub. I only opened the issue from the forum, so I can't do any direct testing myself.

gouchi, 2018-08-08:
We can enable it for Generic platform but for RPi we need to wait we update the kernel to at least 4.10 which should be the case for [Lakka 2.1.2](https://github.com/libretro/Lakka-LibreELEC/issues/515#issuecomment-411457379).

gouchi, 2018-08-08:
@Mr-Jake Can you test this [build](http://nightly.builds.lakka.tv/special_builds/mayflash/) which enable CONFIG_HID_MAYFLASH thanks to @ToKe79.

Mr-Jake, 2018-08-08:
@gouchi Yes, I can test it tonight or tomorrow, and will post the result.

Thanks gouchi and ToKe79.

Mr-Jake, 2018-08-09:
@gouchi I tested the special build you linked me and it works. Lakka sees the Mayflash adapter and controller. I was able to navigate the Lakka menu, configure the buttons, and play in game. I tested both the Mayflash Nintendo adapter and the Mayflash Sega Saturn adapter.

I did notice I had to plug the Gamecube controller into port 2 of the Mayflash adapter to configure the buttons and play in game. Port 1 on the adapter would only navigate the Lakka menus. Not really an issue if the user knows to do this, or perhaps it is my adapter? 

Thanks gouchi and ToKe79. I believe this issue could be closed.

gouchi, 2018-08-09:
@Mr-Jake  Thank you for the feedback. I am glad it is working.

Just for information and if possible can you provide some [log](http://www.lakka.tv/doc/Troubleshooting-Lakka/) especially `lsusb`, `dmesg` output and lakka system log ?

gouchi, 2018-10-10:
Close this issue as PR has been merged.

closed 2018-10-10 by gouchi