libretro/Lakka-LibreELEC #515: Feature request: Add gpio_keys module

Gadgetoid, 2018-08-07:
I'm running Lakka-RPi2.arm-2.1-devel-20180731011450-r28201-g1f26930 on a Raspberry Pi 3B+ and attempting to get Picade HAT up and running to supply on/off button functionality and control input. However the gpio_keys module is missing.

`/lib/modules/<uname>/kernel/drivers/input/keyboard/gpio_keys.ko`

I'm not sure how to submit a PR to include this module, or if it's immediately possible. A quick scan over the config for Raspberry Pi doesn't yield an option to enable it, but it exists upstream in RaspberryPi Linux- https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/input/keyboard/gpio_keys.c

I can currently build/install the dtoverlay with the following commands and the audio, power LED and on/off functionality work:

```
wget https://raw.githubusercontent.com/pimoroni/picade-hat/master/picade.dts -O /tmp/picade.dts
dtc -I dts -O dtb -o /tmp/picade.dtbo /tmp/picade.dts
mount -o remount,rw /flash
cp picade.dtbo /flash/overlays
cp /flash/config.txt /flash/config.txt.beforepicade
echo "dtoverlay=picade" >> /flash/config.txt
mount -o remount,ro /flash
```

kivutar, 2018-08-07:
Is it the module enabled by CONFIG_KEYBOARD_GPIO?

I don't see it in our kernel config https://github.com/libretro/Lakka-LibreELEC/blob/8b8d1c994ae471042c0a0115d0253c81f041b147/projects/RPi2/linux/linux.arm.conf

Maybe because we're still using a 4.9 kernel?

Gadgetoid, 2018-08-08:
Yes, I believe it is, it's upstream in LibreELEC here - https://github.com/LibreELEC/LibreELEC.tv/blob/fe1d2789630eeb5604b8169af52c5c5ba0a63915/projects/RPi/devices/RPi/linux/linux.arm.conf#L1585 (4.14) kernel.

Looking at #470 and other examples of packages, it looks like I could supply a DKMS package for it but I'd rather not if I can help it, since it's already upstream. (and I have enough fingers in enough pies as it is, to be honest!)

Ntemis, 2018-08-08:
Then I will try bumping rpi kernel to 4.14 for Lakka 2.1.2 release

closed 2020-08-18 by Ntemis