libretro/Lakka-LibreELEC #518: Hid sony backport 4.9 for amlogic 3.14

steff75, 2018-08-10:
Hi,

I tried to contact the devs on the [https://forums.libretro.com/t/amlogic-led-controls-and-rumble-dualshock-3-bluetooth/17026](url) about getting the Sony Dualshock 3 fully working on amlogic devices because of missing things on the kernel 3.14, but no dev answered yet.

Now I've seen the latest CoreELEC release, which is a LibreELEC fork with some modifications for amlogic.
They included a port of the ds3/ds4 drivers from the 4.9 kernel. 

I would be very pleased, if you could adopt these changes to Lakka too.
Here's the link:

[https://github.com/CoreELEC/linux-amlogic/pull/8](url)

Many thanks!

Ntemis, 2018-08-10:
what device you have?

steff75, 2018-08-10:
I have an Odroid-C2.

Ntemis, 2018-08-10:
Does the S905 image boot on your board?

Ntemis, 2018-08-11:
i cant switch Odroid-c2 to that kernel but i can try on the S905 platform to do this if you can test.

steff75, 2018-08-11:
Thank you for your efforts. I would like to test, but I have only the Odroid-C2. I already tested the CoreELEC build for my Odroid, and can confirm, that the DualShock backport is working with their build.

I thought, it would be possible to include this into Lakka for all amlogic devices, which use the 3.14 kernel, where the Odroid-C2 is one of them. I'm not sure, if all this devices use the same kernel and the difference is only the dtb file, do they? If that is the case, it would be possible for me to test it on the C2. If not, maybe we can find a tester with a relevant S905 device in the libretro forum.

Ntemis, 2018-08-12:
All S905 devices use the same kernel yes but odroid-c2 has its own so i have to patch that one. I prefer if you can test an S905 image and boots, to fix for all vs just one

steff75, 2018-08-12:
Ok then. Would it be enough, when I switch the kernel file over ssh with your test-kernel and then reboot? Or will there be changes with the other system files too,? Then I will make a test-setup on another sdcard for this.

closed 2020-08-18 by Ntemis