libretro/Lakka-LibreELEC #521: Failure in PPSSPP options

Renetrox, 2018-08-11:
I downloaded version 2.1 of development and ppsspp still has failures. One of them has to do with the option of "frameskip", it is unusable, when setting it, the game disappears, only the sound is heard, when using the option "non buffered" the screen is flipped. I placed the PPSSPP folder in System, but that does not change anything.
Lakka 2.1 nigthlty 31/07/2018
hw: Raspberry pi 3b

kaiomatico, 2019-09-17:
I also have the same Issue on the raspberry pi 4 with every Lakka nightly build (also the latest from today, 17th. September 2019). As soon as Frameskip is enabled heavy flickering or just a black screen occurs. When compiling and running PPSSPP standalone on raspbian desktop (not as a libretro core) frameskip works fine. So it has to be a problem with the libretro port.

natinusala, 2019-09-17:
To be sure can you try the PPSSPP libretro core on raspbian?

It being a libretro issue doesn't surprise me tbh

kaiomatico, 2019-09-17:
Sure I can do that tomorrow. Is it simply an sudo apt install retroarch or do I have to do something else?

natinusala, 2019-09-17:
Building it from source would be the best option

kaiomatico, 2019-09-18:
So that sadly didn't work. I followed this compilation guide: https://docs.libretro.com/development/retroarch/compilation/ubuntu/
./configure showed no errors, I was able to install all -dev libraries. However the compilation with make errors out here:
`CXX obj-unix/release/ui/drivers/qt/options/moc_options.cpp
LD retroarch
/usr/bin/ld: obj-unix/release/gfx/drivers/gl1.o: undefined reference to symbol 'glReadBuffer'
/usr/bin/ld: //usr/lib/arm-linux-gnueabihf/libGLESv2.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make: *** [Makefile:208: retroarch] Error 1
pi@raspberrypi:~/retroarch $ 
`

kaiomatico, 2019-09-18:
Ok so after building retroarch did not work I thought maybe I will just build the ppsspp core myself. so after running ./libretro-fetch.sh ppsspp and then ./libretro-build ppsspp I got the following error:
`make platform="unix-opengl" -j4 CC="gcc" CXX="g++" 
g++ -std=c++11 -O2 -D_NDEBUG  -DVK_USE_PLATFORM_XLIB_KHR -D__LIBRETRO__ -DPPSSPP -DUSE_FFMPEG -DGLEW_STATIC -DGLEW_NO_GLU  -I../ffmpeg/linux/armv7l/include -I. -I.. -I../Common -I../libretro -I../ext/native -I../ext/zlib -I../ext/snappy -I../ffmpeg  -I../ffmpeg/linux/armv7l/include -I../ext/cityhash -I../ext/armips -I../ext/native/ext -I../ext/native -I../ext/libkirk -I../ext/xbrz -I../ext/xxhash -I../ext/glew   -fPIC   -msse -msse2 -D_M_IX86 -D_ARCH_32 -m32 -DHAVE_OPENGL  -DDYNAREC -ffunction-sections -fdata-sections -c -o../ext/snappy/snappy-c.o ../ext/snappy/snappy-c.cpp
g++ -std=c++11 -O2 -D_NDEBUG  -DVK_USE_PLATFORM_XLIB_KHR -D__LIBRETRO__ -DPPSSPP -DUSE_FFMPEG -DGLEW_STATIC -DGLEW_NO_GLU  -I../ffmpeg/linux/armv7l/include -I. -I.. -I../Common -I../libretro -I../ext/native -I../ext/zlib -I../ext/snappy -I../ffmpeg  -I../ffmpeg/linux/armv7l/include -I../ext/cityhash -I../ext/armips -I../ext/native/ext -I../ext/native -I../ext/libkirk -I../ext/xbrz -I../ext/xxhash -I../ext/glew   -fPIC   -msse -msse2 -D_M_IX86 -D_ARCH_32 -m32 -DHAVE_OPENGL  -DDYNAREC -ffunction-sections -fdata-sections -c -o../ext/snappy/snappy.o ../ext/snappy/snappy.cpp
g++ -std=c++11 -O2 -D_NDEBUG  -DVK_USE_PLATFORM_XLIB_KHR -D__LIBRETRO__ -DPPSSPP -DUSE_FFMPEG -DGLEW_STATIC -DGLEW_NO_GLU  -I../ffmpeg/linux/armv7l/include -I. -I.. -I../Common -I../libretro -I../ext/native -I../ext/zlib -I../ext/snappy -I../ffmpeg  -I../ffmpeg/linux/armv7l/include -I../ext/cityhash -I../ext/armips -I../ext/native/ext -I../ext/native -I../ext/libkirk -I../ext/xbrz -I../ext/xxhash -I../ext/glew   -fPIC   -msse -msse2 -D_M_IX86 -D_ARCH_32 -m32 -DHAVE_OPENGL  -DDYNAREC -ffunction-sections -fdata-sections -c -o../ext/xbrz/xbrz.o ../ext/xbrz/xbrz.cpp
g++ -std=c++11 -O2 -D_NDEBUG  -DVK_USE_PLATFORM_XLIB_KHR -D__LIBRETRO__ -DPPSSPP -DUSE_FFMPEG -DGLEW_STATIC -DGLEW_NO_GLU  -I../ffmpeg/linux/armv7l/include -I. -I.. -I../Common -I../libretro -I../ext/native -I../ext/zlib -I../ext/snappy -I../ffmpeg  -I../ffmpeg/linux/armv7l/include -I../ext/cityhash -I../ext/armips -I../ext/native/ext -I../ext/native -I../ext/libkirk -I../ext/xbrz -I../ext/xxhash -I../ext/glew   -fPIC   -msse -msse2 -D_M_IX86 -D_ARCH_32 -m32 -DHAVE_OPENGL  -DDYNAREC -ffunction-sections -fdata-sections -c -o../ext/native/ext/gason/gason.o ../ext/native/ext/gason/gason.cpp
g++: error: unrecognized command line option ‘-msse’; did you mean ‘-fdse’?
g++: error: unrecognized command line option ‘-msse2’
g++: error: unrecognized command line option ‘-msse’; did you mean ‘-fdse’?
g++: error: unrecognized command line option ‘-m32’; did you mean ‘-mbe32’?
g++: error: unrecognized command line option ‘-msse’; did you mean ‘-fdse’?
g++: error: unrecognized command line option ‘-msse2’
g++: error: unrecognized command line option ‘-msse2’
g++: error: unrecognized command line option ‘-m32’; did you mean ‘-mbe32’?
g++: error: unrecognized command line option ‘-m32’; did you mean ‘-mbe32’?
make: *** [Makefile:438: ../ext/snappy/snappy-c.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Makefile:438: ../ext/xbrz/xbrz.o] Error 1
g++: error: unrecognized command line option ‘-msse’; did you mean ‘-fdse’?
g++: error: unrecognized command line option ‘-msse2’
g++: error: unrecognized command line option ‘-m32’; did you mean ‘-mbe32’?
make: *** [Makefile:438: ../ext/native/ext/gason/gason.o] Error 1
make: *** [Makefile:438: ../ext/snappy/snappy.o] Error 1
cp "ppsspp_libretro.so" "/home/pi/libretro-super/dist/unix/ppsspp_libretro.so"
cp: cannot stat 'ppsspp_libretro.so': No such file or directory
1 core(s) failed:
   ppsspp
`

I guess this is because gcc / g++ is used while ppsspp standalone requires cmake / clang?

natinusala, 2019-09-18:
Yeah ppsspp uses cmake now. To fix retroarch you can add --disable-qt to ./configure

You may be able to grab prebuilt binaries on the buildbot if building from source still fails

kaiomatico, 2019-09-18:
I can't find a prebuilt retroarch binary and also no ppsspp core here: http://buildbot.libretro.com/nightly/linux/armhf/latest/

Am I missing the correct link? Thank you in advance.

Renetrox, 2019-09-18:
The only current solution to this failure is to use an earlier version of ppsspp. In raspberry pi 3 the version that has no flaws in the videos or frameskip is 1.01

Renetrox, 2019-09-18:
https://tgho.st/nbZJMeGUa7

closed 2020-08-18 by Ntemis