libretro/Lakka-LibreELEC #528: Xbox one controller doesn't work on rock64

JoyBurrito, 2018-08-30:
The sysrem can recognize xbox controller with terms shown as '100% microsoft x-box one s pad configured in port #0', but the buttons don't work.

- Which version of Lakka are you using? 2.1.1 stable
- What system hardware are you using?rock64
- What did you do? Plug a xboxone controller into the usb port and try to control the system. 
- What did you expect to happen? Xbox controller should be able to control the lakka system.
- What happened instead? After a vibration, the light on controller didn't turn on and nothing happened when pressing buttons.

I'll borrow a keyboard trying to get the log tomorrow...
Please help

RobLoach, 2018-08-30:
I've found similar issues with udev on Linux. Two questions...

1. What input driver are you on?
2. Mind saving the autoconfig and uploading the file here? Then we can confirm it against https://github.com/libretro/retroarch-joypad-autoconfig .

JoyBurrito, 2018-08-30:
Thank you for your kindness. In fact I am a layman with linux, flashing the img file using etcher the software. Don't know what is the input driver. I'll try to fetch the config files, may taking quite some time.

RobLoach, 2018-08-31:
[image]

JoyBurrito, 2018-08-31:
God, why there is no driver option on my system?

RobLoach, 2018-08-31:
Delete retroarch.cfg and load it up again? What is the default one that shows up.

JoyBurrito, 2018-09-02:
Another issue, i changed the driver settings and many other settings then reboot the system. Found the setting option on the main manu gone missing...

JoyBurrito, 2018-09-02:
Then i tried putty to connect my system, key in input_driver = "udev".The report shows"-sh:input driver:not found"

JoyBurrito, 2018-09-02:
[lakka config.zip](https://github.com/libretro/Lakka-LibreELEC/files/2343217/lakka.config.zip)
I uploaded .config and etc these two files. Is this helpful?

JoyBurrito, 2018-09-08:
I tried to add microsoft xbox one s pad.cfg to lakka /joypads/. But after a reboot, whole system lost video signal.

JoyBurrito, 2018-09-09:
In fact, i tried flashing ubuntu to my rock64 and installed xboxdrv driver, also succeeded starting xboxdrv.service. But in jstest, as previous, pressing buttons did nothing to this system.

closed 2020-08-18 by Ntemis